teslamotors/vehicle-command #2: Unlatch Model 3 doors

prodriguezdlt, 2023-10-11:
Will there be a command in the future to unlatch the doors? Thanks o/

ArchGryphon9362, 2023-10-11:
You just simply send a closure move request of type open to the necessary door over BLE and that unlatches!

thefireblade, 2025-01-22:
The implementation is currently not on FleetAPI but the pieces of code that you need to implement this exists in the Vehicle Command protocol: https://github.com/teslamotors/vehicle-command/blob/07123bf4e03444538a62361f790fd7e6da835ab5/pkg/vehicle/vcsec.go#L226

From my surface level inspection, it seems like the to implement this is checking the protobuf for ClosureMoveRequest https://github.com/teslamotors/vehicle-command/blob/07123bf4e03444538a62361f790fd7e6da835ab5/pkg/protocol/protobuf/vcsec.proto#L93-L102

And then adding your own custom actions: https://github.com/teslamotors/vehicle-command/blob/main/pkg/vehicle/actions.go#L12-L29

closed 2025-01-22 by thefireblade

stx, 2025-01-28:
@thefireblade The car firmware appears to be preventing action on the doors.