teslamotors/vehicle-command #3: iOS app

prodriguezdlt, 2023-10-11:
Is it possible to create an iOS app with this and still execute BLE commands?

prodriguezdlt, 2023-10-17:
> There shouldn't be anything stopping you from doing this. Although, currently Tesla requires you to provide which URLs are going to be using the API, and you'll likely need to proxy the requests through a server of your own.

How would i create an ios app with this golang framework? is it even posible to use golang on an xcode project? Thanks in advance

sethterashima, 2023-10-19:
[Gomobile](https://pkg.go.dev/golang.org/x/mobile/cmd/gomobile) generates bindings to turn a Go package into an iOS/Android library. You'd need to implement the [Connector interface](https://pkg.go.dev/github.com/teslamotors/vehicle-command@v0.0.1/pkg/connector#Connector) using native BLE functions, and then use that connector when calling NewVehicle ([example](https://github.com/teslamotors/vehicle-command/blob/main/examples/ble/main.go#L68)). The first two bytes of each BLE message encode the remaining length of that message ([example](https://github.com/teslamotors/vehicle-command/blob/main/pkg/connector/ble/ble.go#L100-L101)).

I'll look into pulling the Tesla-specific logic out of the [BLE connector package](https://pkg.go.dev/github.com/teslamotors/vehicle-command@v0.0.1/pkg/connector/ble) so it can be re-used by third-party apps more easily.

JSpiner, 2023-10-30:
I'm trying to develop unofficial sdk in multiple languages/platforms.
Currently, the fleet API is implemented with python, and I'm trying to develop the ble API with kotlin(for Android).
If you are interested in iOS native development, please contribute.

https://github.com/JSpiner/TeslaSdk

prodriguezdlt, 2023-10-30:
> I'm trying to develop unofficial sdk in multiple languages/platforms. Currently, the fleet API is implemented with python, and I'm trying to develop the ble API with kotlin(for Android). If you are interested in iOS native development, please contribute.
> 
> https://github.com/JSpiner/TeslaSdk

I would totally pay you some money to implement BLE over iOS native, we can exchange emails or discord user if you d like o/

georgevicbell, 2023-11-14:
I'm confused how this would work...sharing the private key via an app would be problematic...

itsMeDavidV, 2023-11-30:
> [Gomobile](https://pkg.go.dev/golang.org/x/mobile/cmd/gomobile) generates bindings to turn a Go package into an iOS/Android library. 

@sethterashima running into issues with this, specifically because GoMobile skips a lot of methods for lack of supported param/return types.

See below some examples of this in pkg/account after conversion:

```
// skipped method Account.Get with unsupported parameter or return types

// skipped method Account.GetVehicle with unsupported parameter or return types

// skipped method Account.Post with unsupported parameter or return types

// skipped method Account.SendVehicleFleetAPICommand with unsupported parameter or return types
```

// skipped method Account.UpdateKey with unsupported parameter or return types

GeorgeBellTMH, 2023-12-08:
Was thinking today it might be helpful if there was a quick way to get this up and running on AWS lambda...low cost and doesn't require the key being shipped with the app...

georgevicbell, 2023-12-08:
Was thinking it might be good to have a quick way to spin up an AWS lambda for this or something...low cost and doesn't require shipping the secrets...

sheng168, 2024-02-19:
I've written enough Swift code to add key, perform handshake and lock/unlock using BLE only

prodriguezdlt, 2024-02-19:
> I've written enough Swift code to add key, perform handshake and lock/unlock using BLE only

Will you be willing to share it man? cheers o/

sheng168, 2024-02-28:
Possibly, what are you trying to do? I've been busy trying to get it release to the App Store. It's basically the same code running in both iOS and watchOS. However watchOS has very low connection limits that are causing problems.

prodriguezdlt, 2024-02-28:
> Possibly, what are you trying to do? I've been busy trying to get it release to the App Store. It's basically the same code running in both iOS and watchOS. However watchOS has very low connection limits that are causing problems.

I have a custom private app that integrates some of my intelligent home devices and i want to integrate some of the stuff for my tesla, so i basically want a demo ios app that connects to my tesla, i can unlatch doors and do regular stuff but with BLE functionality, and then i would just basically integrate that code into my custom app.

sheng168, 2024-03-02:
What kind of api would you like to see? Would you be willing to help build it?

thefireblade, 2025-01-22:
> [Gomobile](https://pkg.go.dev/golang.org/x/mobile/cmd/gomobile) generates bindings to turn a Go package into an iOS/Android library. You'd need to implement the [Connector [...]

Closing old stale issue with this

closed 2025-01-22 by thefireblade