teslamotors/vehicle-command #5: [Document Request] Request to add ble specification document.

JSpiner, 2023-10-13:
I couldn't believe my eyes when I saw this great project that released the code to control a car with ble.
There is huge potential for use. Thank you very much.
I am curious if there are any plans to create a separate document for the specifications for controlling vehicles with BLE.
The devices on which Linux-based golang ble can run are limited.
It would be of great help if documentation was provided to support implementation on Android or iOS, which are used by many people.

sheng168, 2024-02-19:
I have implemented an iOS BLE client base on information in this repo, running tesla-control -debug ...
and decoding the TX/RD hex dumps and reproducing them

sethterashima, 2024-03-08:
https://github.com/teslamotors/vehicle-command/blob/main/pkg/protocol/protocol.md

closed 2024-03-08 by sethterashima