teslamotors/vehicle-command #6: registration fails

rawmean, 2023-10-13:
Registering the public ket fails with http error 400 and this: 
`{"response":null,"error":"Invalid domain: https://www.myDomain.com","error_description":""}`

```
curl --header 'Content-Type: application/json' \
  --header "Authorization: Bearer $TESLA_API_TOKEN" \
  --data '{"domain":"https://www.myDomain.com"}' \
  'https://fleet-api.prd.na.vn.cloud.tesla.com/api/1/partner_accounts' 
```
This is a bottleneck without it nothing else works. 
My guess is that this has to do with the way the server expects the public key being hosted at:

`https://www.myDomain.com/.well-known/appspecific/com.tesla.3p.public-key.pem` 
Can you please give an example?

Has anybody been able to register the public key?

ncristovao, 2023-10-13:
I'm having the same issue as well. I was able to register as a developer with Tesla and get an access token, but the request to the partner_accounts API results in the same error you're getting; "Invalid domain."

Has anyone been able to successfully register?

agbpatro, 2023-10-13:
some insights as to why it could be a erroring out:-
- Domain should be valid and registered to you
- The domain name shouldn't start with `https://`
- Domain name should be lowercase. Something like `mydomain.com`

ncristovao, 2023-10-14:
> some insights as to why it could be a erroring out:-
> 
> * Domain should be valid and registered to you
> * The domain name shouldn't start with `https://`
> * Domain name should be lowercase. Something like `mydomain.com`

Thanks, I tried this, but still without success. When registering on developer.tesla.com, https:// is required for the domain. And when POSTing to this `https://fleet-api.prd.na.vn.cloud.tesla.com/api/1/partner_accounts` endpoint, whether I try with or without https://, I receive the same invalid domain error.

Any other ideas? Were you able to successfully register?

rawmean, 2023-10-14:
> some insights as to why it could be a erroring out:-
> 
> * Domain should be valid and registered to you
> * The domain name shouldn't start with `https://`
> * Domain name should be lowercase. Something like `mydomain.com`

I have tried with and without `https://`, with and without `www.`, and with all lower-case. 
I still get the same error. Have you actually been able to register the public key successfully?

aaronpkahn, 2023-10-16:
@ncristovao @rawmean what error response code and response body are you receiving?  (please don't include any sensitive details)

tomhollander, 2023-10-17:
I'm having the same problem. A few thoughts:
 - The docs don't specify what to do with the private key. Surely it must be used at some point the process?
 - The website used in this registration process doesn't seem to have anything to do with the app URLs. Maybe there's a requirement that the referrer or reply URLs are hosted on the same domain? (Given you can't edit these URLs in an existing app it's a bit hard to test).

rawmean, 2023-10-17:
Seems like Tesla has fixed the problem with registration on Oct 16

tomhollander, 2023-10-17:
> Seems like Tesla has fixed the problem with registration on Oct 16

Hmm it's still not working for me. Can you share your request payload?

aaronpkahn, 2023-10-17:
@tomhollander Good feedback. We'll update the docs at https://developer.tesla.com/docs/fleet-api#register to better explain why registering a public key is necessary and where the private key is used: it's loaded into the vehicle command proxy to sign commands.

As for domain requirements, the root domain (second-level + top-level) must match a domain from your allowed_origins on developer.tesla.com

ncristovao, 2023-10-17:
> @tomhollander Good feedback. We'll update the docs at https://developer.tesla.com/docs/fleet-api#register to better explain why registering a public key is necessary and where the private key is [...]

Thanks for the reply. In the Authentication section, I noticed a typo that could cause a little confusion.

In the 'Generating a third-party token on behalf of a customer' under Parameters, the scope parameter description says 
'office_access' instead of 'offline_access'.

It would also be great if the next section on how to use the code from the callback to generate a refresh token had some more details in step 4. It has the code to the right on how to generate new access tokens, which is helpful, but it would be nice if there was a parameters section similar to the step above that outlined which params were required and the value to use.

Is it also possible to list the official TTL of access tokens and refresh tokens?

Thanks for your responses!

rawmean, 2023-10-17:
> > Seems like Tesla has fixed the problem with registration on Oct 16
> 
> 
> 
> Hmm it's still not working for me. Can you share your request payload?

Make sure you don't include "https://". It's too bad that the documentation does not explain this. 
I used the exact curl command that it's in the documentation

rawmean, 2023-10-17:
> @ncristovao @rawmean what error response code and response body are you receiving?  (please don't include any sensitive details)

The error code was 400 (BYW, I responded to tesla api fleet support a few days ago who asked the same question, but received no follow up). 

The problem was resolved yesterday without me making any changes so someone must have fixed it on the server side.

tomhollander, 2023-10-17:
> @tomhollander Good feedback. We'll update the docs at https://developer.tesla.com/docs/fleet-api#register to better explain why registering a public key is necessary and where the private key is [...]

Thanks for the response @aaronpkahn. Requiring the domain to match the allowed origin makes a lot of sense, but I didn't see this anywhere in the documentation. It would be good to update the description of the `partner_accounts` endpoint to make this clear, as well as the need to not include the https:// in the payload.  

While you're here, a couple of related questions:
 - It isn't apparently possible to edit or delete existing app registrations - even just during testing my apps are piling up. Are you planning to address this?
 - The energy APIs aren't currently documented - when can we expect these to be in Fleet API?

rawmean, 2023-10-18:
Tesla has fixed this. Registration works now

closed 2023-10-18 by rawmean

rawmean, 2023-10-18:
> 
> * It isn't apparently possible to edit or delete existing app registrations - even just during testing my apps are piling up. Are you planning to address this?
> * The energy APIs aren't currently documented - when can we expect these to be in Fleet API?

Please open a new issue for topics that are not related to "registration failure" (which has been fixed already).
I have opened one for energy endpoints not being documented. Thanks for brining that up.