teslamotors/vehicle-command #9: When creating a third-party token on behalf of a client, a server_error response was received

Wiks00, 2023-10-16:
Greetings.
Recently set up a 3rd party account in FleetAPI and tried to go through the authentication procedure to generate a client token, but encountered a problem at the stage of exchanging the authentication code for the access token and refresh token pair. 

The response  
`{
  "error": "server_error",
  "error_description": "Internal server error"
}`

This could be because we are not planning to generate a partner authentication token and hence skipping the step of generating the public/private key pair for signing commands ?

Note: we are using deeplink as redirect_uri

And an additional question: what is the purpose of redirect_uri in post /token endpoint? [Per RFC 6749](https://www.rfc-editor.org/rfc/rfc6749#section-3) there is no mention of POST /token requiring redirects.

Regards.

vbarrier, 2023-10-16:
From the documentation, partner token generation and create a public key is mandatory to generate an client access token.

Wiks00, 2023-10-16:
> From the documentation, partner token generation and create a public key is mandatory to generate an client access token.

Went through the public key registration process and still get the same error...

vbarrier, 2023-10-16:
Do you use the same domain as registered for your app?

To be sure it is the response when you ask a token with authorization code given in the redirect_uri?

Wiks00, 2023-10-17:
The public key is hosted in the same domain that is specified in the allowed origin at the time of registration. But we use a mobile application for authentication. 

Yes, that is right, this is the response to exchanging code for an access token.

vbarrier, 2023-10-17:
Maybe you can add details with your appId / url used in order to someone look into.

Wiks00, 2023-10-18:
The problem was that the /token endpoint expects the same redirect_url that was specified in /authorize