teslamotors/vehicle-command #11: Ability to edit redirect_uri / redirect_url

vbarrier, 2023-10-17:
Hi

I didn’t found any way to edit app settings from the developer.tesla.com website

I would like to add localhost url or the default from postman from testing oauth: https://oauth.pstmn.io/v1/browser-callback

andig, 2023-10-26:
Currently no way to do so. Quickly setting up a test environment, that's an absolute show stopper.

andig, 2023-10-28:
ping @teslamotors this is an absolute show-stopper.

andig, 2023-11-03:
Friendly ping @teslamotors @tesla-sdk-publish there are already >20 upvotes. Is there any chance to get this addressed short-term? We'd need to double- and triple-register otherwise for testing and production changes would not be possible.

JSpiner, 2023-11-03:
If it is difficult to implement an edit function, how about creating a delete function first? Currently, tesla developer console can only create two apps, so it is difficult to test them anymore.

aaronpkahn, 2023-11-03:
We'll be adding the ability to edit redirect_uri, return_url, and allowed_origins in the coming weeks.  Sorry for the lame inconvenience/blocker.

rawmean, 2023-11-07:
> We'll be adding the ability to edit redirect_uri, return_url, and allowed_origins in the coming weeks. Sorry for the lame inconvenience/blocker.

Is it also possible to allow editing scope? Users will be prompted to approve/deny the new scope.
If one forgets to add a scope a new domain needs to be acquired and registered and a new https certificate needs to be acquired and it's a huge hassle. Not to mention that users need to re-log in.

Thanks

grantmac, 2023-11-16:
Any timeline on this @aaronpkahn ?

grantmac, 2023-11-17:
It also looks like you cannot register more than 5 apps...

JSpiner, 2023-11-18:
@grantmac yes it is. So I created an additional new account.
Sadly this is the only solution.

jbanyer, 2023-11-27:
This appears to be possible now. I just edited the redirect URLs for an existing app.

Kelex60fps, 2023-11-27:
How can I create a redirect link pls

ThomasAlxDmy, 2024-06-25:
Yup you can do it now via the UI.

andig, 2024-06-25:
Plus: you can even changes scopes now 👍🏻

closed 2024-06-25 by sethterashima