teslamotors/vehicle-command #13: Why does registration always fail?

Zephery, 2023-10-18:
This error always appears during the final submission.
[image]

How to fix this?

vbarrier, 2023-10-18:
Hello, check in the developer console you can get details

vbarrier, 2023-10-18:
I had this error because I already have a Tesla business account associated to the company details I put to create the app settings. 

So I ended up to use the same Tesla account to create the App settings.

vbarrier, 2023-10-18:
🥇

jshay21, 2023-10-18:
I tried for the first time to create an app last night and got the instant "Rejected" notification

vbarrier, 2023-10-18:
> > Hello, check in the developer console you can get details
> 
> @vbarrier How can we get into the developer console without being able to create an app?

Developer console of your browser and you will get the "real" http error with a text detail. In my case it was: "An account already exist with that company"

Zephery, 2023-10-19:
Same errors! I really don’t know how to register!
Help~

gak, 2023-11-11:
Has anyone solved this? I always get an instant rejection.

rickychu, 2023-11-14:
I got instant rejection if I include localhost in any of the links.

Zephery, 2023-11-26:
"Unable to Onboard" means tesla api is not support in some area! So I buy a American server, it work!

Zephery, 2023-11-26:
[image]
@vbarrier Sorry to bother you, Did you successfully registered the application from developer.tesla.com?
I don't know how to proceed with this step. Does it mean I have to develop on devices with Bluetooth？
```text
./tesla-control -ble add-key-request public_key.pem owner cloud_key
Error: failed to find a BLE device: can't init hci: no devices available:
```
Do you have any suggestions? Thanks a lot!

2Fast2BCn, 2024-03-28:
[image]

[image]

whatever that means?

eTaphee, 2024-03-30:
> ![image](https://private-user-images.githubusercontent.com/4680358/317683930-e94246aa-6d94-4c66-a343-817f5ac64e62.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3L [...]

I got the same response.

My region is Korea.

Chacsam, 2024-04-12:
> ![image](https://private-user-images.githubusercontent.com/4680358/317683930-e94246aa-6d94-4c66-a343-817f5ac64e62.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3L [...]

Same response here, Belgium.
Would that be the reason?

doordiey, 2024-05-14:
> ![image](https://private-user-images.githubusercontent.com/4680358/317683930-e94246aa-6d94-4c66-a343-817f5ac64e62.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3L [...]

I got the same response.

jmcollin78, 2024-06-29:
FYI, it have been fixed now.

2Fast2BCn, 2024-07-01:
Not for me.