teslamotors/vehicle-command #15: Energy endpoints are not documented

rawmean, 2023-10-18:
The old endpoints for energy do not work (with the new base URL) and the new ones are not documented.
For example `/api/1/energy_sites/\(siteID)/history` does not work with the Fleet API

jbanyer, 2023-11-15:
Plus one for this. Our system [chargehq.net] uses both vehicle data and energy data (Powerwall / solar). We can't migrate our users to the new API until it supports the energy endpoints.

Currently we rely on the following endpoints for energy data:
- `GET /api/v1/products` - to obtain Battery and Solar products
- `GET /api/1/energy_sites/${energy_site_id}/live_status` - to obtain live power data

tomhollander, 2023-11-15:
I think the `/products` endpoint works now, but I couldn't find any energy endpoints.

andig, 2024-01-05:
Seems they are now: https://developer.tesla.com/docs/fleet-api#energy-endpoints

closed 2024-01-07 by rawmean