teslamotors/vehicle-command #16: Protobuf sources

gak, 2023-10-20:
Just wondering if it's possible to include the proto sources in this repository?

I ask in case someone wants to port this code to another language, they would need to generate protobuf sources from the original `.proto` files.

Cheers

acvigue, 2023-11-15:
Tried my best to recreate equivalent `.proto` source files from the Go code, feel free to take a look and let me know if you have any suggestions!

https://github.com/acvigue/TeslaProtobufs

gak, 2023-11-15:
Nice work! Cheers for making it open source. I can't try it because #13 :(

acvigue, 2023-11-16:
I am using it with BLE but yeah couldn't get fleet api to work for same reason. Just pairing keys to car manually and controlling it

itsMeDavidV, 2023-11-30:
> Just wondering if it's possible to include the proto sources in this repository?

I see that there is the endpoint: [/api/1/vehicles/{id}/signed_command](https://developer.tesla.com/docs/fleet-api#signed_command) and one of the params for it is a Base64 encoded Tesla Vehicle Command Protocol message. 

Though it would not be ideal, as an alternative to using this SDK could mobile devs could we just use OG protobufs with this [endpoint](https://developer.tesla.com/docs/fleet-api#signed_command)?

sethterashima, 2023-12-06:
Added in https://github.com/teslamotors/vehicle-command/pull/86.