teslamotors/vehicle-command #17: `upstream internal error` at every api call

JSpiner, 2023-10-21:
I'm trying to use the api with partner authentication.

https://auth.tesla.com/oauth2/v3/token 
The access_token was generated successfully using the above API.
However, when I try to use any API, the following error occurs.

ex)
```
$ curl --location --request GET 'https://fleet-api.prd.na.vn.cloud.tesla.com/api/1/vehicles' \
--header 'Authorization: Bearer {my access token} ' 
```

```json
{
    "response": null,
    "error": "upstream internal error",
    "error_description": ""
}
```
This happens the same in all APIs.

I registered the public key needed for partner authentication. at here : https://tesla-api.jspiner.io/.well-known/appspecific/com.tesla.3p.public-key.pem
And, I also called the registration API. 
I didn't note the response at that time, but if I call it again now, it says it has already been registered.
```json
POST https://fleet-api.prd.na.vn.cloud.tesla.com/api/1/partner_accounts
{
    "response": null,
    "error": "Validation failed: Public key hash has already been taken",
    "error_description": ""
}
```

vbarrier, 2023-10-25:
I think it does work for partner accounts at the moment

JSpiner, 2023-10-30:
solved it.
The problem is access key that I used was partner access token. 
When I use 3rd party access token, it works.

close #17

nbosio1001, 2023-11-02:
I created a 3rd party token. Do I use that 3rd party token with the list vehicle endpoint? I tried that so far and it returns nothing. I am confused on how using a 3rd party access token is connected to your account.

JSpiner, 2023-11-02:
@nbosio1001 

When a user with a Tesla account is authenticated through oauth, a 3rd party token is created. You can view the list of vehicles of users authenticated with this token. If I authenticate in my account with an oauth link containing your app key, my vehicle will be searched.

In my experience, if empty data is returned, it's a scope(permission) issue. Test it by requesting with the minimum scope.
Tesla API is still unfriendly to failure cases.

JSpiner, 2023-11-02:
Thinking about it again, in the situation in the text, it seems normal for an error saying 'accessToken is incorrect' to appear, but it seems abnormal for an 'upstream internal error' to appear. So reopen it.

drnic, 2023-11-16:
Perhaps remove the section on generating a partner auth token if its not going to be supported

https://developer.tesla.com/docs/fleet-api#generating-a-partner-authentication-token

drnic, 2023-11-16:
I am getting the 500 error for the partner specific `public_key` endpoint too https://developer.tesla.com/docs/fleet-api#partner-endpoints

thefireblade, 2025-01-22:
This is unusual but a lot of changes has been made since, I am going to close this old issue and filed an internal ticket. Feel free to reopen a new issue if this is still occurring

closed 2025-01-22 by thefireblade