teslamotors/vehicle-command #18: A `redirect_uri` error occurred during 3rd party authentication.

JSpiner, 2023-10-21:
I am trying to do 3rd party authentication.
However, redirect_uri is not recognized in the first step of oauth.
This is my request

https://auth.tesla.com/oauth2/v3/authorize?client_id={my_client_id}&prompt=login&redirect_uri=tesla-api.jspiner.io/oauth/redirect&response_type=code&scope=openid vehicle_device_data vehicle_cmds vehicle_charging_cmds user_data offline_access&state=ababab

```json
{
    "error": {
        "message": "We don't recognize this redirect URI/URL. Please use the redirect URI/URL found in your app details.",
        "code": 400,
        "reason": "AuthorizationError",
        "referenceID": "1e4bff194e894618a80439498b20549a-1697875590936"
    }
}
```

In my developer dashboard, redirect_uri is `https://tesla-api.jspiner.io`
So I tested the URLs below. But they all fail with the same error.

- tesla-api.jspiner.io
- https://tesla-api.jspiner.io
- http://tesla-api.jspiner.io
- https://tesla-api.jspiner.io/oauth/redirect

cwardcode, 2023-10-24:
+1, having the same issue here.  I emailed the fleetapisupport email last week and haven't heard anything back...

akitathai94, 2023-10-28:
{
  "error": {
    "message": "We don't recognize this redirect URI/URL. Please use the redirect URI/URL found in your app details.",
    "code": 400,
    "reason": "AuthorizationError",
    "referenceID": "41a3e44f1b324c07b9bf37b72bd2917e-1698524849019"
  }
}

JSpiner, 2023-10-30:
@cwardcode @akitathai94 
I solved it. 
The uri I registered in console and used in api is like this : `https://tesla-api.jspiner.io`
It works if I use a uri that includes the path like `https://tesla-api.jspiner.io/callback`.
In my opinion, in Tesla server internal code, uri path is non-null type.
Although it has been resolved, I believe this is still a problem and I will leave the issue open.

special thanks to @yangs1202 for advice.

cwardcode, 2023-10-31:
It's still not working from my end - for me I'm attempting to call the endpoint from a web frontend, and forming the address such as:

> https://auth.tesla.com/oauth2/v3/authorize/?client_id=<my-registered-client-id>&redirect_uri=https://<my-registered-url>&response_type=code&scope=openid%20offline_access%20vehicle_device_data%20vehicle_cmds%20vehicle_charging_cmds&nonce=a58c20e404e911f608ab7f93d99d5ac5cfQBXZe0X&state=3523415bae84f3c7917125910ce349759077raMA2

If I attempt to login, I get the Invalid Redirect URI error. Attempting to login using `https://<my-registered-url>/callback` does not seem to work either.

Did you have to re-create your registration and include a redirect path with a return type? Not sure why it isn't working for me.

JSpiner, 2023-10-31:
@cwardcode 

What did you register in the Tesla console?
If you specify a new path after the registered URL, it will fail.
You must enter exactly the same text as the redirect url specified when creating the app in the console.

grantmac, 2023-11-16:
@JSpiner 

I assume you created a new app to set your new [domain]/callback URI?

JSpiner, 2023-11-16:
@grantmac 
you're right. There is no function to modify the Tesla console yet.

cwardcode, 2023-11-17:
@JSpiner  That seems to be my issue, my Redirect URUI didn't have a path, only the base domain, and because I've already registered the domain, it won't allow me to register a new app. Since I can't change the redirect URI through their site, I'm stuck waiting 😢

JSpiner, 2023-11-18:
>  @JSpiner That seems to be my issue, my Redirect URI didn't have a path, only the base domain, and because I've already registered the domain, it won't allow me to register a new app using that website. Since I can't change the redirect URI through their site, I'm stuck waiting 😢

Yes same happens to me. You need to create new domain. However, even if you don't create a completely new domain, just change the A record and it will be recognized as a different domain.

ex) tesla-api.ABCD.com -> tesla-new-api.ABCD.com : it works!
ex) tesla-api.ABCD.com -> tesla-api.ABCD.com/newApi :  not works

cwardcode, 2023-11-19:
Oh wow, that's interesting! I thought it was actual domain validation, and not only at the sub-domain level. Its interesting they chose this technique. Thank you for the information!!

JSpiner, 2023-11-19:
@cwardcode 
Glad you found it helpful. 
If you are interested, please also visit my project to library Tesla API into multiple languages.

https://github.com/JSpiner/TeslaSdk

thefireblade, 2025-01-22:
Closing this old issue, seems to be resolved and documentation improvements have been made since

closed 2025-01-22 by thefireblade