teslamotors/vehicle-command #19: Vehicle endpoints /list and /vehicle return `null` for property `option_codes`

Darko, 2023-10-23:
Hey!

When fetching a vehicles list, or a single vehicle from the `GET /api/1/vehicles/{id}` or `GET /api/1/vehicles` the property `option_codes` is returned with a `null` value.

Is this intentional, and if so, which endpoint should be used to fetch the option codes for a car?

rawmean, 2023-10-24:
They stopped populating `option_codes` correctly even in the legacy API. Use vehicle_config instead

aaronpkahn, 2023-10-24:
@Darko that is confusing; this is a legacy field, and we'll remove it.  There is a new option codes endpoint coming in the next few weeks.

Darko, 2023-10-24:
@aaronpkahn that is awesome! Is there a particular discord channel or somewhere else where we can monitor change logs?

itsMeDavidV, 2023-10-24:
> There is a new option codes endpoint coming in the next few weeks.

Much needed, thank you!

aaronpkahn, 2023-10-27:
For now we'll post updates here: https://developer.tesla.com/docs/fleet-api#announcements-amp-api-changelog, or if we move to somewhere else we'll link from the documentation

Darko, 2023-11-21:
@aaronpkahn Any update on this? It's been a few weeks? 😅

Whirlwind, 2024-07-25:
It still not works in china developer website.

thefireblade, 2025-01-22:
> For now we'll post updates here: https://developer.tesla.com/docs/fleet-api#announcements-amp-api-changelog, or if we move to somewhere else we'll link from the documentation

We have an internal ticket to fix this, closing with this but for now, just ignore

closed 2025-01-22 by thefireblade