teslamotors/vehicle-command #20: {"response":null,"error":"upstream internal error","error_description":""} for public key and weird error for register. What am I doing wrong?

nbosio1001, 2023-10-23:
I use this command to get public keys
`TESLA_API_TOKEN="<Bearer token from partner authentication token>"
curl --header 'Content-Type: application/json' \
  --header "Authorization: Bearer $TESLA_API_TOKEN" \
  'https://fleet-api.prd.na.vn.cloud.tesla.com/api/1/partner_accounts/public_key'`
but I get this error `{"response":null,"error":"upstream internal error","error_description":""}`
I also try the register curl command, but it tells me to remove https://, I remove https://, and leave fleet-api... and it stays on bash register.sh.

JSpiner, 2023-10-24:
https://github.com/teslamotors/vehicle-command/issues/17 same here

nbosio1001, 2023-11-01:
> #17 same here
Are you still having the same problem?
{"response":null,"error":"Invalid domain: string. Ensure that domain is lowercase and starts without https://","error_description":""}
I am also having this issue

closed 2023-11-02 by nbosio1001