teslamotors/vehicle-command #21: App not recognize anymore

vbarrier, 2023-10-24:
I'm trying to get a new partner token (it worked before)

and now I'm getting: 

`{
    "error": "unauthorized_client",
    "error_description": "We don't recognize this client_id and client_secret combination",
    "error_uri": "https://auth.tesla.com/error/reference/9c0edd7f-ca24-49ab-a2e9-cab13cbb7ee1-1698135037237"
}`

Darko, 2023-10-25:
Make sure to register on both the NA and EU regions. If you've done it only on one of them, you won't be able to create and use tokens for other regions. So just register for all regions.

vbarrier, 2023-10-25:
Thank you for your reply.

How do I register on both? Should I create 2 apps?

closed 2023-10-25 by vbarrier

Darko, 2023-10-25:
With the same app client id and secret, make a request to the dedicated region api for registration. For example if you did a POST to NA:
`POST https://fleet-api.prd.na.vn.cloud.tesla.com/api/1/partner_accounts`
send a post request to EU api as well
`POST https://fleet-api.prd.eu.vn.cloud.tesla.com/api/1/partner_accounts`