teslamotors/vehicle-command #28: dial tcp: lookup fleetapi-europe.vn.teslamotors.com: no such host

andig, 2023-11-03:
Registered app and user created token against app. Using token to do a 

	account, err := account.New(token.AccessToken)
	if err != nil {
		return nil, err
	}

	if _, err := account.Get(ctx, "api/1/vehicles"); err != nil {
		return nil, err
	}

fetching vehicles fails:

    error fetching api/1/vehicles: Get "https://fleetapi-europe.vn.teslamotors.com/api/1/vehicles": dial tcp: lookup fleetapi-europe.vn.teslamotors.com: no such host

rawmean, 2023-11-06:
I get the same error for EU cars. My guess is that when you registered your app with Tesla, you did not ask for the `user_data ` scope. I think this scope is necessary to get the `/api/1/users/region` to return the Base URL. 

Changing the scope for an app is not possible now. This means that another domain needs to be acquired and registered with Tesla, etc. 

This is just a guess. someone please correct me if you think my guess is incorrect. I hope I'm wrong because I don't want to re-do all the work associated with registering a new domain.

andig, 2023-11-07:
> I get the same error for EU cars. My guess is that when you registered your app with Tesla, you did not ask for the user_data  scope. I think this scope is necessary to get the /api/1/users/region to return the Base URL.

I did actually. But what server I'm direct to MUST not depend on which scopes I have requested. Much more, the target server should exist ;)

@aaronpkahn could you kindly check what's wrong with Europen vehicles/ server configuration?

rawmean, 2023-11-09:
> I did actually. But what server I'm direct to MUST not depend on which scopes I have requested. Much more, the target server should exist ;)
>
> @aaronpkahn could you kindly check what's wrong with [...]

That's true. I am surprised that more people haven't paid attention to this non-existent url

ArchGryphon9362, 2023-11-09:
> Registered app and user created token against app. Using token to do a
> 
> ```
> account, err := account.New(token.AccessToken)
> if err != nil {
> 	return nil, err
> }
> 
> if _, err := account.Get(ctx, "api/1/vehicles"); err != nil {
> 	return nil, err
> }
> ```
> 
> The user's token has the following audiences:
> 
> ```
> https://fleet-api.prd.na.vn.cloud.tesla.com
> https://fleet-api.prd.eu.vn.cloud.tesla.com
> https://auth.tesla.com/oauth2/v3/userinfo
> ```
> 
> Although account is in Europe, vehicle-command choses `https://fleet-api.prd.na.vn.cloud.tesla.com` as domain. Then, fetching vehicles fails (notice different domain):
> 
> ```
> error fetching api/1/vehicles: Get "https://fleetapi-europe.vn.teslamotors.com/api/1/vehicles": dial tcp: lookup fleetapi-europe.vn.teslamotors.com: no such host
> ```
> 
> If I change `acount.go` to pick `https://fleet-api.prd.eu.vn.cloud.tesla.com`, request fails with
> 
> ```
> http error when sending command to https://fleet-api.prd.eu.vn.cloud.tesla.com/api/1/vehicles: 412 Precondition Failed
> ```

Well the first error is just an invalid redirect (mistake on Tesla’s part), but the second error of 412 is a mistake on your part. It’s reaching the Tesla servers this time, but your partner account is unregistered. This is a documented error in the Fleet API reference.

If everything is done right (and you contact the correct region), the first error (with the invalid redirect) shouldn’t even matter, and Tesla could just replace it with a “404 vehicle not found in this region” or something.

andig, 2023-11-09:
> Well the first error is just an invalid redirect (mistake on Tesla’s part), but the second error of 412 is a mistake on your part. It’s reaching the Tesla servers this time, but your partner account is unregistered. This is a documented error in the Fleet API reference.

Though so, but indeed it is registered and public key reachable. Anyway, this is OT. This issue is about invalid tesla host name.

rawmean, 2023-11-10:
When I force it to sent to `https://fleet-api.prd.eu.vn.cloud.tesla.com` (for an EU car), it fails and wants to retransmit to: `https://fleetapi-america.vn.teslamotors.com` which also does not exists.

rawmean, 2023-11-15:
I just noticed that Tesla has changed the non-exiting URL to one that actually exists. But the problem is that the command still does not execute. Here is an example:

```
2023-11-15T23:04:07Z [info ] Received POST request for /api/1/vehicles/<vin>/command/set_charge_limit
2023-11-15T23:04:07Z [debug] Sending request to https://fleet-api.prd.na.vn.cloud.tesla.com/api/1/vehicles/<vin>/signed_command: {"routable_message":""}
2023-11-15T23:04:07Z [debug] Server returned 421: Misdirected Request: {"response":null,"error":"user out of region, use base URL: https://fleet-api.prd.eu.vn.cloud.tesla.com, see https://developer.tesla.com/docs/fleet-api#regional-requirements","error_description":""}
2023-11-15T23:04:07Z [warn ] [07294242a4b2b24fc91408c972aad021] Terminal transmission error: {"response":null,"error":"user out of region, use base URL: https://fleet-api.prd.eu.vn.cloud.tesla.com, see https://developer.tesla.com/docs/fleet-api#regional-requirements","error_description":""}
2023-11-15T23:04:07Z [error] Returning error Misdirected Request
2023-11-15T23:04:07Z [debug] [7e94c382321fc10f951ffcdff77d7172] Retrying transmission after error: Post "https://fleet-api.prd.na.vn.cloud.tesla.com/api/1/vehicles/<vin>/signed_command": context canceled
```

Summary:
- It sends to the NA server first
- It fails (because it is an EU car)
- Retries transmission to and fails again

Why can't it send the command to the correct server URL to begin with?

andig, 2023-11-16:
@rawmean which audience are you using when creating the 3rd party token? Docs only mention NA, are you using EU server url?

Urkman, 2023-11-27:
Did someone managed to send a command to a EU vehicle?
I used the eu in every step and I'm able to get data using the vehicle_data endpoint.

I get this error:

`Error: {"response":null,"error":"user out of region, use base URL: https://fleet-api.prd.eu.vn.cloud.tesla.com, see https://developer.tesla.com/docs/fleet-api#regional-requirements","error_description":""}`

andig, 2023-11-27:
Did you also sign up the partner account in EU? That seemed to do the trick for me, but I‘m not entirely sure afterwards.

Urkman, 2023-11-27:
Yes, I did... And I can access the vehicle data, but not end a command...

andig, 2023-11-27:
…and you‘ve patched the library to always send to the EU endpoint?

Urkman, 2023-11-27:
No, I patched nothing...

jbanyer, 2023-11-28:
> I get the same error for EU cars. My guess is that when you registered your app with Tesla, you did not ask for the `user_data ` scope. I think this scope is necessary to get the `/api/1/users/region` to return the Base URL.

For the record, it is [no longer] necessary to have the `user_data` scope to call the `/api/1/users/region` endpoint. The [docs](https://developer.tesla.com/docs/fleet-api#region) say it works with any of these:

> scopes: vehicle_device_data, vehicle_cmds, energy_cmds, energy_device_data, user_data

andig, 2023-11-28:
> No, I patched nothing...

I've patched the library to chose the EU endpoint instead of taking the first one from the audiences.

rawmean, 2023-11-28:
> For the record, it is [no longer] necessary to have the `user_data` scope to call the `/api/1/users/region` endpoint. The [docs](https://developer.tesla.com/docs/fleet-api#region) say it works with any of these. I can confirm it works without the `user_data` scope.
> 
> >

If the `region` endpoint is to be used for determining the base url, how come it requires us to know the base URL for the user? It seems like we need to know the base url (ie, na vs. eu) to send the `region` query: 

```
curl --header 'Content-Type: application/json' \
  --header "Authorization: Bearer $TESLA_API_TOKEN" \
  'https://fleet-api.prd.na.vn.cloud.tesla.com/api/1/users/region' 
```

jbanyer, 2023-11-28:
> If the `region` endpoint is to be used for determining the base url, how come it requires us to know the base URL for the user? It seems like we need to know the base url (ie, na vs. eu) to send the `region` query:

A quick test shows that the EU `region` endpoint will successfully return the region for an NA user:
```
GET https://fleet-api.prd.eu.vn.cloud.tesla.com/api/1/users/region
{
  response: {
    region: 'na',
    fleet_api_base_url: 'https://fleet-api.prd.na.vn.cloud.tesla.com'
  }
}
```
So it appears you can use any region to make the `region` call. I haven't tried the other user endpoints.

Urkman, 2023-11-29:
> I haven't tried the other user endpoints.

I can't make a call to the `users/me` Endpoint. With this error: `Unauthorized missing scopes`. But I added the `user_data` scope to the authorization request.

I also wrote a mail to the support, but no answer. 
I think, that the new API still needs a lot of work and we don't get any answer from Tesla about the problems here.
The only thing they do: Reject calls to several cars and tell uns to move to the (not 100% working) new API.

jbanyer, 2023-12-05:
> I've patched the library to chose the EU endpoint instead of taking the first one from the audiences.

@andig any chance you could publish your patch? My Go is very rusty :)

andig, 2023-12-05:
@jbanyer after https://github.com/teslamotors/vehicle-command/blob/main/pkg/account/account.go#L117 I'm just forcing the Domain to EU:

	domain = "fleet-api.prd.eu.vn.cloud.tesla.com"

closed 2023-12-06 by sethterashima

andig, 2023-12-07:
Confirmed working, thank you @sethterashima