teslamotors/vehicle-command #31: How to determine if the car has installed our public key

rawmean, 2023-11-05:
Is there a way to determine if the car has installed our public key?
We can open `https://tesla.com/_ak/<your_domain_name>` the first time the user logs in but a user may have more than one car. So, we need to open this url multiple times (once for each car).

sethterashima, 2023-11-13:
Currently, the best option might be to attempt to call [StartSession](https://pkg.go.dev/github.com/teslamotors/vehicle-command@v0.0.1/pkg/vehicle#Vehicle.StartSession).

I could speak to the mobile app team about adding a callback; e.g., the mobile app would POST the VIN to some path on `<your_domain_name>`. Would that work for you?

stx, 2023-11-13:
@sethterashima I don't think a POST callback on install would work - since the key can be revoked/deleted later, you'd lose the status of it, so services need some way to test for the existence of the key.

Can that StartSession be invoked via the HTTP proxy?

I think we need "https://localhost:4443/api/1/vehicles/$id/installed_keys" or "https://localhost:4443/api/1/vehicles/$id/start_session" or something similar.

rawmean, 2023-11-15:
> Currently, the best option might be to attempt to call [StartSession](https://pkg.go.dev/github.com/teslamotors/vehicle-command@v0.0.1/pkg/vehicle#Vehicle.StartSession).
> 
> This doesn't seem like a very elegant solution for your needs, though. I could speak to the mobile app team about adding a callback; e.g., the mobile app would POST the VIN to some path on `<your_domain_name>`. Would that work for you?

Thanks or the response. Is it possible to add an endpoint which allows us to query the car to determine whether the car has our public key? POSTing to our domain has the problem that if the user deletes the public key, we would not be notified.

To clarify: I need to know if the car has already installed the public key so that my app doesn't keep asking the user to install the key. Currently, I have to ask the user to confirm that they have done it already.

rawmean, 2023-11-21:
> Currently, the best option might be to attempt to call [StartSession](https://pkg.go.dev/github.com/teslamotors/vehicle-command@v0.0.1/pkg/vehicle#Vehicle.StartSession).
>
> This doesn't seem like a [...]

When a client sends a command through the proxy to a car that has not installed the public key associated with the app, proxy issues the following:
```
2023-11-21T01:44:20Z [debug] Server returned 200: OK: {"response":”MhNHMTx1...AhADkgMQIWPUIskQbJg=="}
2023-11-21T01:44:20Z [error] Returning error Internal Server Error
```

This means that even in the case that the car returns a “Internal Server Error” (because it has not installed the public key, the client receives a 'http 200’ response. Therefore, the client has no way of knowing if the command failed because the car has not installed the public key; it just knows that the proxy sent the command to the car.

rawmean, 2023-11-21:
> Can that StartSession be invoked via the HTTP proxy?

The proxy does invoke `[car.StartSession](https://github.dev/teslamotors/vehicle-command/blob/e1ca5088e3c55f85c7c3f251dd306181520e4afc/pkg/proxy/proxy.go#L268)`, but the problem is that the client will never see the result of this call failing. The proxy just logs the event (as shown in my previous comment above).

jbanyer, 2023-11-21:
@rawmean when I send a command (eg `set_charging_amps`) via the proxy to a car without my public key, I get this response:
```
{
   "http_code":500,
   "reply":{
      "response":null,
      "error":"vehicle rejected request: your public key has not been paired with the vehicle",
      "error_description":""
   }
}
```

So it appears possible to detect that a vehicle does not have the key installed, although it requires sending a command.

rawmean, 2023-11-22:
> So it appears possible to detect that a vehicle does not have the key installed, although it requires sending a command.

You are absolutely right. Thank you for correcting me.
This is a good workaround but, as you mentioned, is not ideal because requires sending a command and having it fails to determine if the pub key has been installed.

simon-norman, 2024-01-02:
Just to verify it's safe to use StartSession to check that the car is connected (IE not actually sending any commands, just to verify), e.g. we don't need to do any cleanup, close the session etc.?

simon-norman, 2024-01-05:
Hi @sethterashima might you be able to advise on this?

simon-norman, 2024-01-09:
@sethterashima sorry just to confirm this is safe to do? Any cleanup we need to do after calling startsession?

sethterashima, 2024-03-08:
Cleaning out older tickets, apologies for not responding to the question.

It's safe to use StartSession to check if a key is paired.

You should call vehicle.Disconnect to free client resources.

closed 2024-03-08 by sethterashima