teslamotors/vehicle-command #37: send signed commands without using this sdk

jbanyer, 2023-11-15:
I understand that this repo contains code which can generate signed vehicle commands, and that the `tesla-http-proxy` application will convert "old" REST requests into these signed commands before sending to the vehicle.

Is it possible to generate signed commands without using this sdk or the proxy?

The API docs describe a [signed_command](https://developer.tesla.com/docs/fleet-api#signed_command) endpoint, but it links to this repo.

My system is developed in node.js / javascript, and it might be nice to avoid having to add the proxy component to the architecture, unless there is some other good motivation to use it?

eiannone, 2023-12-23:
Hi @jbanyer , I'm doing exactly this. I wrote the fleet api client in node.js, and it's working. I can already send commands that don't require signing, like "sessionInfoRequest".
Currently I'm writing the message signing part, and I believe it should work. I will publish the code on github when ready.

RonnyWinkler, 2023-12-28:
Hello @jbanyer @eiannone,
please add some comments or links to a Java/JS/NodeJS implementation or some examples how to prepare the encrypted body for the signed message.
I want to implement the signed message into a NodeJS app to execute the request directly without the proxy (that's really buggy and partially returning 500/Timeout errors), but i'm still searching for documentations or examples.
Many thanks.

eiannone, 2023-12-31:
Hi @RonnyWinkler , I've just published my code here: [https://github.com/eiannone/tesla-cmd-api](https://github.com/eiannone/tesla-cmd-api).
This is an initial draft, but the message signing works. It's a node.js port of original Go code, but very stripped down. I have extracted only the message signing part and the Fleet API calls.
For now I've implemented only the session start command (which doesn't require signing) and the chargingSetLimit command (which does). 
Starting from `chargingSetLimit()` method in `CarServer `class, it should be very easy to add new commands, just looking in the `car_server.proto` file.

RonnyWinkler, 2023-12-31:
Hi @eiannone ,
many thanks for your work. That's a nice start into 2024 🎆
I'll take a look into next days.

sethterashima, 2024-01-18:
Protocol spec added in https://github.com/teslamotors/vehicle-command/pull/142. Feedback is welcome.

I haven't looked at your implementation, @eiannone, but please look over the spec since it contains some information on some checks that are necessary for security but not functionality.

closed 2024-01-18 by sethterashima

jonahwh, 2024-02-20:
I managed to put together a Kotlin SDK (https://github.com/boltfortesla/tesla-fleet-sdk-kotlin) using the documentation. Found one minor issue and proposed #182  to fix it