teslamotors/vehicle-command #38: We can't edit our Credentials/API . I have to keep submitting a new request and it automatically gets denied

onlyusedtesla, 2023-11-15:
Please let me know when you will have the Edit Credentials / API . i have been emailing the fleet email no reply for 2 weeks. 
I have to submit a new app request which keeps getting automatically denied. Thank you.

fyllepo, 2023-11-15:
Same issue here, we've been struggling for days.

andig, 2023-11-16:
See https://github.com/teslamotors/vehicle-command/issues/11#issuecomment-1797080505. No point in double-posting ;)

onlyusedtesla, 2023-11-27:
We are getting enter VALID URL ( Credentials)  error and we generated from AuthO. Can you help us diagnose ?

onlyusedtesla, 2023-11-28:
seems like sub-domains disallowed?

thefireblade, 2025-01-22:
Resolved with https://github.com/teslamotors/vehicle-command/issues/11#issuecomment-2188118841

closed 2025-01-22 by thefireblade