teslamotors/vehicle-command #39: EU fleet vehicles can no longer be controlled because SDK doesn't send commands to EU vehicles

stx, 2023-11-16:
I'm adding a new issue for visibility and clarity since this is affecting production now; an increasing number of EU fleet vehicles can no longer be controlled.

Appears to be the same underlying issue as: https://github.com/teslamotors/vehicle-command/issues/36 https://github.com/teslamotors/vehicle-command/issues/28 https://github.com/teslamotors/vehicle-command/issues/32

An increasing number of vehicles on Fleet API now return this 403 error:

`{"response":null,"error":"Tesla Vehicle Command Protocol required, please refer to the documentation here: https://developer.tesla.com/docs/fleet-api#2023-10-09-rest-api-vehicle-commands-endpoint-deprecation-warning","error_description":""}`

The command SDK is automatically routing requests to the NA region only. Proxy logs:

```
2023-11-16T10:19:55-08:00 [debug] Creating proxy
2023-11-16T10:19:55-08:00 [info ] Listening on localhost:4443
2023-11-16T10:19:59-08:00 [info ] Received POST request for /api/1/vehicles/<EU_VIN_HERE>/command/lock
2023-11-16T10:19:59-08:00 [debug] Executing lock on <EU_VIN_HERE>
2023-11-16T10:19:59-08:00 [info ] Starting dispatcher service...
2023-11-16T10:19:59-08:00 [info ] Requesting session info from DOMAIN_VEHICLE_SECURITY
2023-11-16T10:19:59-08:00 [info ] Requesting session info from DOMAIN_INFOTAINMENT
2023-11-16T10:19:59-08:00 [debug] Sending request to https://fleet-api.prd.na.vn.cloud.tesla.com/api/1/vehicles/<EU_VIN_HERE>/signed_command: {"routable_message":"..."}
2023-11-16T10:19:59-08:00 [debug] Sending request to https://fleet-api.prd.na.vn.cloud.tesla.com/api/1/vehicles/<EU_VIN_HERE>/signed_command: {"routable_message":"..."}
2023-11-16T10:19:59-08:00 [debug] Server returned 421: Misdirected Request: {"response":null,"error":"user out of region, use base URL: https://fleet-api.prd.eu.vn.cloud.tesla.com, see https://developer.tesla.com/docs/fleet-api#regional-requirements","error_description":""}
2023-11-16T10:19:59-08:00 [warn ] [16a4ac6e06ab2b878c182cdfded3b98b] Terminal transmission error: {"response":null,"error":"user out of region, use base URL: https://fleet-api.prd.eu.vn.cloud.tesla.com, see https://developer.tesla.com/docs/fleet-api#regional-requirements","error_description":""}
2023-11-16T10:19:59-08:00 [error] Returning error Misdirected Request
2023-11-16T10:19:59-08:00 [debug] Server returned 421: Misdirected Request: {"response":null,"error":"user out of region, use base URL: https://fleet-api.prd.eu.vn.cloud.tesla.com, see https://developer.tesla.com/docs/fleet-api#regional-requirements","error_description":""}
2023-11-16T10:19:59-08:00 [warn ] [3cdd90b572f39393e1ee908d204af6f4] Terminal transmission error: {"response":null,"error":"user out of region, use base URL: https://fleet-api.prd.eu.vn.cloud.tesla.com, see https://developer.tesla.com/docs/fleet-api#regional-requirements","error_description":""}
```

The partner account is set up in both regions and the Fleet API is fully working for both regions.

The failure appears due to the following:

In the access token, the NA region appears first:

```
  "aud": [
    "https://fleet-api.prd.na.vn.cloud.tesla.com",
    "https://fleet-api.prd.eu.vn.cloud.tesla.com",
    "https://auth.tesla.com/oauth2/v3/userinfo"
  ],
```

In account.go, it takes the first one (NA) here: https://github.com/teslamotors/vehicle-command/blob/e1ca5088e3c55f85c7c3f251dd306181520e4afc/pkg/account/account.go#L82-L100C2

```
func (p *oauthPayload) domain() string {
        for _, u := range p.Audiences {
                if strings.HasPrefix(u, "https://auth.tesla.com") {
                        continue
                }
                d, _ := strings.CutPrefix(u, "https://")
                d, _ = strings.CutSuffix(d, "/")
                if !domainRegEx.MatchString(d) {
                        continue
                }
                if remapped, ok := remappedDomains[d]; ok {
                        return remapped
                }
                if strings.HasSuffix(d, ".tesla.com") || strings.HasSuffix(d, ".tesla.cn") {
                        return d
                }
        }
        return ""
}
```

rawmean, 2023-11-17:
duplicate #28

jbanyer, 2023-12-04:
Has anyone successfully sent signed commands to an EU vehicle using the proxy? Has anyone published a patched version of the proxy which supports both regions?

closed 2023-12-06 by sethterashima