teslamotors/vehicle-command #40: vehicle rejecting owners api commands - deprecated?

jbanyer, 2023-11-16:
Has the old "owners" API been deprecated for some vehicles?

I am seeing this error for a small number of vehicles (currently at least 4 vehicles):

```
POST https://owner-api.teslamotors.com/api/1/vehicles/<REDACTED>/command/set_charging_amps 20
Response code 403 (Forbidden)
{
  response: null,
  error: 'Tesla Vehicle Command Protocol required, please refer to the documentation here: https://developer.tesla.com/docs/fleet-api#2023-10-09-rest-api-vehicle-commands-endpoint-deprecation-warning',
  error_description: ''
}
```

All impacted vehicles have:
- software version: 2023.32.100 2b8c8db7ccb3
- api_version: 63

However, we have many vehicles with the same software version where commands are working normally.

stx, 2023-11-16:
Same issue. Particularly problematic for EU vehicles since the command protocol doesn't work there (https://github.com/teslamotors/vehicle-command/issues/39).

itsMeDavidV, 2023-11-17:
@TeslaMotors people, would it be possible to have a bit more of a detailed timeline on the deprecation of the old REST API?

This would be greatly appreciated :)

jbanyer, 2023-11-19:
I've now seen this error with different software versions, including:
- 2023.38.6 c1f85ddb415f
- 2023.32.9 757be7f0a072

So far around 0.5% of vehicles connection to our system are impacted.

jbanyer, 2023-11-23:
For those who missed it (I did until now), Tesla have announced the deprecation timeline:

https://developer.tesla.com/docs/fleet-api#2023-11-17-rest-api-vehicle-commands-endpoint-deprecation-timeline-action-required

<p style="box-sizing: border-box; outline-offset: 2px; max-inline-size: var(--tds-max-paragraph-width); font-size: 14px; line-height: 1.6; margin-right: 699.5px; padding: 0px 28px; display: block; margin-top: 0px; color: rgb(51, 51, 51); font-family: &quot;Gotham SSm&quot;; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; white-space: normal; background-color: rgb(255, 255, 255); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;">Please migrate any command API integrations to the Tesla Vehicle Commands protocol using the<span> </span><a href="https://github.com/teslamotors/vehicle-command" style="box-sizing: border-box; outline-offset: 2px; background-color: transparent; color: inherit;">SDK</a>.</p>
Dates | Change
-- | --
November 2023 | Newly delivered vehicles* will only support the Tesla Vehicle Command protocol after this date
Nov - Dec 2023 | REST API support will be deprecated on existing customer vehicles that have not used the REST API in the preceding 30 days
January 2024 | All vehicles* will require Tesla Vehicle Command protocol. The REST API will be fully deprecated

<p style="box-sizing: border-box; outline-offset: 2px; max-inline-size: var(--tds-max-paragraph-width); font-size: 14px; line-height: 1.6; margin-right: 699.5px; padding: 0px 28px; display: block; margin-top: 0px; color: rgb(51, 51, 51); font-family: &quot;Gotham SSm&quot;; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; white-space: normal; background-color: rgb(255, 255, 255); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;">*Fleet accounts are excluded from these changes until further notice</p><p style="box-sizing: border-box; outline-offset: 2px; max-inline-size: var(--tds-max-paragraph-width); font-size: 14px; line-height: 1.6; padding-block-start: 14px; margin-right: 699.5px; padding: 0px 28px; display: block; margin-top: 0px; color: rgb(51, 51, 51); font-family: &quot;Gotham SSm&quot;; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; white-space: normal; background-color: rgb(255, 255, 255); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;">When issuing a REST API command to a vehicle that requires the Tesla Vehicle Command protocol, Fleet API will return the following error: "Tesla Vehicle Command Protocol required, please refer to the documentation here: https://developer.tesla.com/docs/fleet-api#2023-10-09-rest-api-vehicle-commands-endpoint-deprecation-warning (403)"</p>Please migrate any command API integrations to the Tesla Vehicle Commands protocol using the [SDK](https://github.com/teslamotors/vehicle-command).

Dates	Change
November 2023	Newly delivered vehicles* will only support the Tesla Vehicle Command protocol after this date
Nov - Dec 2023	REST API support will be [deprecated](https://developer.tesla.com/docs/fleet-api#2023-10-09-rest-api-vehicle-commands-endpoint-deprecation-warning) on existing customer vehicles that have not used the REST API in the preceding 30 days
January 2024	All vehicles* will require Tesla Vehicle Command protocol. The REST API will be fully deprecated
*Fleet accounts are excluded from these changes until further notice

When issuing a REST API command to a vehicle that requires the Tesla Vehicle Command protocol, Fleet API will return the following error: "Tesla Vehicle Command Protocol required, please refer to the documentation here: https://developer.tesla.com/docs/fleet-api#2023-10-09-rest-api-vehicle-commands-endpoint-deprecation-warning (403)"

closed 2023-11-23 by jbanyer