teslamotors/vehicle-command #43: Command SDK doesn't recognize window_control

stx, 2023-11-17:
**window_control** is missing from this switch statement, which causes window controls to work through Fleet API but not the SDK:

https://github.com/teslamotors/vehicle-command/blob/main/pkg/proxy/execute.go#L110

```

Fleet API:

curl -H "Authorization: Bearer <token>" -H "Content-Type: application/json" --data '{"command":"vent"}' https://fleet-api.prd.na.vn.cloud.tesla.com/api/1/vehicles/<vehicle_id>/command/window_control

-> {"response":{"result":true,"reason":""}}

Command SDK:

curl -H "Authorization: Bearer <token>" -H "Content-Type: application/json" --data '{"command":"vent"}' https://localhost:4443/api/1/vehicles/<vin>/command/window_control

-> {"response":null,"error":"invalid_command","error_description":""}

```

rawmean, 2023-11-17:
Add this to `pkg/proxy/execute.go` after line 419

```
	case "window_control":
		cmd, err := params.getString("command", true)
		if err != nil {
			return err
		}
		if cmd == "vent" {
			return car.VentWindows(ctx)
		} else {
			return car.CloseWindows(ctx)
		}
```

ststolz, 2024-03-06:
Fix for current main: 

https://github.com/teslamotors/vehicle-command/compare/main...ststolz:ststolz-tesla-vehicle-command:main

thefireblade, 2025-01-22:
closing old issue, the code has been update since

closed 2025-01-22 by thefireblade