teslamotors/vehicle-command #44: Charge invoice PDFs

grantmac, 2023-11-17:
The GET /api/1/dx/charging/history endpoint gives us access to charging history.

In the response is:
```
"invoices": [
    {
        "fileName": "XXXXXXXXX-GB.pdf",
        "contentId": "e32ec779-78ee-490c-bce6-XXXXXXX",
        "invoiceType": "IMMEDIATE"
    }
],
```
Can we access those PDFs?

corsair, 2023-12-03:
The [Discovery Tier](https://developer.tesla.com/docs/fleet-api#membership-tiers:~:text=Enterprise%20charging%20API-,Invoices,-Limited%20Fleet%20Telemetry) currently lists "Invoices" as a provided capability, however no documentation is provided or made available to download these customer PDFs. This gives the perception that this is intended for us to access.

The Fleet API support team communicated that invoices are "[...] only able to be downloaded by Tesla for Business (T4B) accounts or otherwise via the official mobile app", however this directly contradicts the existence of such capabilities by the third-party service/app "Standard Fleet", which allows for the requesting and downloading of charge invoice PDFs on both personal accounts and T4B-delegated accounts.

Providing this functionality to all third-party services ensures equal and fair access to the same [advertised features](https://developer.tesla.com/docs/fleet-api#membership-tiers:~:text=Enterprise%20charging%20API-,Invoices,-Limited%20Fleet%20Telemetry), and is also a crucial offering expected by many vehicle owners utilizing such services.

sapkra, 2024-03-01:
It's now documented how to access the PDFs: https://developer.tesla.com/docs/fleet-api?shell#charging_invoice

So this can be closed.

thefireblade, 2025-01-22:
Closing old issue