teslamotors/vehicle-command #45: Leased cars cannot add third-party apps

rawmean, 2023-11-17:
It is quite usual in EU/UK for people not to have access to owner credentials when they lease a car. This makes it impossible for them to add a 3rd-party app. They get a message like this from Tesla app (translated): "You are not authorized to share this vehicle. Contact the owner of the vehicle to grant third-party access."

Specifically, this happens after the 3rd-party app launches Tesla app to have it install the public key associated with the app on the car.

Is this going to change?

jbanyer, 2023-11-21:
I can confirm the same restriction applies to a DRIVER (not OWNER) account for an Australian vehicle.

[image]

I hope this will be changed. A person with DRIVER access is able to perform most (all?) commands using the Tesla app, so it makes sense to me that they can also perform commands via a third-party app.

rawmean, 2023-11-22:
> I hope this will be changed. A person with DRIVER access is able to perform most (all?) commands using the Tesla app, so it makes sense to me that they can also perform commands via a third-party app.

Especially since a Driver can send commands to the car using the Tesla app.

vbarrier, 2023-11-25:
How can someone have an owner access from Tesla for Business management website? 

From my T4B account I can only assign drivers access... And my T4B admin account isn't not even an owner...

That said, It seems that any car linked to a T4B account can't use third-party apps (commands)

jbanyer, 2023-12-29:
Any update here? It appears there is no solution for fleet vehicles (tesla for business), ie there is no way to install a virtual key on those vehicles.

itsMeDavidV, 2023-12-30:
So for clarification, you're saying that anyone with a leased vehicle currently can't enroll a 3rd party public key?

rawmean, 2024-01-07:
> So for clarification, you're saying that anyone with a leased vehicle currently can't enroll a 3rd party public key?

Depends on the car owner/dealer. In Europe more dealers that lease the car, do not give the Owner Credentials for people who lease the car. So, this because a problem if the person who leases a car wants to use a 3rd party app.

jbanyer, 2024-01-22:
We received this statement from Tesla (fleetapisupport@tesla.com) which suggests that a fix is coming for this:

> The ability for lessees/drivers to grant permissions to third party applications and services will be available with an upcoming Tesla App update. Once released the driver will be able to confirm permission sharing using their Tesla vehicle touchscreen and keycard.

markglh, 2024-01-31:
Has anyone received the app update yet or know of an ETA? Kinda blocked from charging using intelligent tariffs at cheap rates until this is fixed.

dapper27, 2024-02-03:
Appears that app version 4.29.5 (iOS) has fixed this. I've just downloaded it but not actually gone through the whole process yet. After trying to add a virtual key it says you need to use your keycard so presumably need to be in the car.

markglh, 2024-02-03:
I can confirm this works on ios now. Still broken on android though

closed 2024-02-04 by rawmean

fmvdb, 2024-02-04:
What would be needed to make it work on android phones?

jbanyer, 2024-02-04:
I can also confirm it works on iOS and not yet on Android. Perhaps this ticket could be re-opened until Android is working.

fmvdb, 2024-02-08:
Today I noticed a new version of the Tesla app on android. This new version fixed this! 
You need to be in the car with your keycard and confirm the access on the screen of the Tesla.

Hejmn, 2024-02-09:
is anyone having issues with the actual in car steps? 
We are experiencing when the keycard is placed on the center console as prompted to by app, clicking approve on the screen, nothing happens. No success message on the Tesla screen, or anyting in the Tesla App.

feyazbek, 2024-02-10:
Hello. I'm not sure if i did the right thing or not. Here is what I have: 
1- Model 3 Not leasing. it is working fine with the command API "Tesla Vehicule Command Proxy"
2- model Y 2023 leasing: does not work with the "Tesla Vehicule Commad Proxy" but works with the REST-API. 
3- model X 2023 leasing: does not work with the "Tesla Vehicule Commad Proxy" but works with the REST-API.  

Is that the way it is supposed to work or am I missing some setting so I can get the "Tesla Vehicule Command Proxy" to work?

stevenlenders, 2024-02-11:
I just followed the procedure after my tesla app on my android was updated to 29.5 and initially I got the same error, but then used the button to do it with the key card and it worked. Tested from my teslafi account.

Underground56, 2024-06-12:
Hi there,
I'm driving a leased 2023 Model Y and I'm trying to setup teslemetry for Solar charging. Unfortunately with latest IOS Tesla App I'm not able to setup access rights for Teslemetry correctly. I can read car data, but not send any command to the car. On the teslemetry console, "Access Type: DRIVER and Command Sign: REQUIRED" is displayed, however in the tesla APP Teslemetry is setup with full access rights in the Security menü. Anyone having a hint whats wrong? I really would love to use Teslemery

best regards
Michael

Underground56, 2024-06-15:
I want to use Teslemetry with a leased Model Y. I was able to setup the access for Teslemetry, both in the car an also in the Tesla App Teslemetry is configured and listed as accessible. However, in the Teslemetry console, the status is still 
Access Type:DRIVER
Command Sign:REQUIRED 
[image]
[image]

Does anyone know how to solve this?

phulstaert, 2024-06-16:
Model Y - Leased Car
Android app version 4.33.5-2673
Car software: 2024.20.1

Not able to add public key

[image]

ViNYLeq, 2024-07-15:
> Model Y - Leased Car Android app version 4.33.5-2673 Car software: 2024.20.1
> 
> Not able to add public key
> 

Were you able to fix this issue? I have same situation.

stevenlenders, 2024-07-16:
Yes, with the new app version, you follow the process in the car. At a certain moment you need to present your key card to proceed.

ViNYLeq, 2024-07-16:
This is not required, I have finished my config late night without a need to go and use the card.

mangio85, 2025-01-09:
did you manage to solve it? if so how..? Thank you very much.

phulstaert, 2025-01-09:
I honestly stopped trying to reach the vehicle locally. I just use the fleet API for most of my requests.
I am super happy with my Tesla, but this is a read dark spot...

stevenlenders, 2025-01-09:
Yes, for me it was solved with a new version of the Tesla app

marine1988, 2025-01-17:
Hello I have android and is 2025 anyway of make this work I tried in October 2024 no luck there