teslamotors/vehicle-command #46: Proposal: allow reading private key from []bytes or io.Reader

andig, 2023-11-18:
Currently this is only possible from filename which is highly inconvenient and cannot be worked around due to the internal packages used.

sethterashima, 2023-12-02:
Is this what you're looking for: https://pkg.go.dev/github.com/teslamotors/vehicle-command@v0.0.1/pkg/protocol#UnmarshalECDHPrivateKey?

A pull request for an `io.Reader` variant would be welcome.

andig, 2023-12-02:
Indeed, thank you. Not being a crypto guy I'm wondering why the implementation is so much different from `LoadExternalECDHKey`? Does this need just the raw bytes without any PEM etc encoding?
From user perspective (e.g. setting keys as ENV, it would seem nice to have something like `LoadExternalECDHKey` that includes the entire encoding as well.

closed 2024-03-08 by sethterashima