teslamotors/vehicle-command #48: What are session best practices?

andig, 2023-11-18:
Sending commands to vehicle requires running session. A session can only be started if the vehicle is awake apparently. That seems to imply that sessions may be short-lived, i.e. should only be established as long as they are needed for running a command.

Is that the best practice?

jbanyer, 2023-11-28:
> It is also not clear if `Wakeup` requires a an established session or if wakeup requests even need by signed?

I have observed that `wake` commands are not signed by the proxy - they are forwarded directly the the API. They also always send the same reply, whether the vehicle is awake or not, and whether they work or not.

andig, 2023-11-28:
@jbanyer OT: seems it would be nice to have a developers forum around the new API as we're all facing the same issues. Any idea if such exists?

sethterashima, 2023-12-02:
- Establishing a session will not keep the vehicle awake.
- A session only expires when the infotainment system or vehicle security controller reboots. If a client attempts to use an expired session, the vehicle responds with a handshake message to start a new session. When this happens, the SDK automatically resets the session and reattempts the command.
- Wakeup requires a session over BLE but not over the Internet. (See https://github.com/teslamotors/vehicle-command/pull/54).

A client can save/road sessions from disk to save a round trip for the handshake. See https://pkg.go.dev/github.com/teslamotors/vehicle-command@v0.0.1/pkg/cache for more information.

andig, 2023-12-02:
@sethterashima highly appreciated. Would be great to add that to the docs. I assume wakeup not needing a session also doesn‘t need it be signed?

sethterashima, 2023-12-06:
Wakeup over the Internet does not need to be signed (since a sleeping car can't authenticate the command, this avoids a chicken-and-egg problem). Wakeup over BLE _does_ need to be signed (the vehicle security controller is always awake, and the command tells it to wake up the infotainment system).

closed 2023-12-06 by sethterashima