teslamotors/vehicle-command #49: Unable to call /api/1/users/me as user_data scope missing from access token

drnic, 2023-11-18:
When I generate an access token I am asking for the `user_data` scope, which is required for `/api/1/users/me` and other endpoints. But this scope is not being included in the access token, and so I cannot get the details of the authenticated/authorized user. Any seen this and have ideas what I might be missing?

```bash
#!/bin/bash

# scopes we want, which includes user_data
# unrelated: offline_access is provided here, which is mentioned in docs, but it too is not in access token
scope="openid user_data offline_access vehicle_device_data vehicle_cmds vehicle_charging_cmds"
: ${TESLA_CLIENT_ID:?required}
: ${TESLA_CLIENT_SECRET:?required}
AUDIENCE=${AUDIENCE:-https://fleet-api.prd.na.vn.cloud.tesla.com}

# if access_token.txt doesn't exist then:
if [[ -f access_token.txt ]]; then
  echo "access_token.txt exists so using it"
  echo
else

  token_response=$(
    curl -sS 'https://auth.tesla.com/oauth2/v3/token' \
      -H 'Content-Type: application/json' \
      -d "$(jq -n \
        --arg client_id "$TESLA_CLIENT_ID" \
        --arg client_secret "$TESLA_CLIENT_SECRET" \
        --arg code "${code:?required}" \
        --arg scope "$scope" \
        '{
            grant_type: "authorization_code",
            client_id: $client_id,
            client_secret: $client_secret,
            code: $code,
            scope: $scope,
            audience: "https://fleet-api.prd.na.vn.cloud.tesla.com",
            redirect_uri: "https://rails-9292.drnicwilliams.com/auth/tesla/callback"
          }')"
  )
  access_token=$(echo "$token_response" | jq -r .access_token)
  # id_token=$(echo "$token_response" | jq -r .id_token)

  echo "access_token: ${access_token}"
  echo $access_token | jwt decode -
  echo

  echo "Saving access_token to access_token.txt"
  echo "$access_token" >access_token.txt
fi

# the following fails because access token does not have `user_data` scope, even though we asked for it:
curl "${AUDIENCE}/api/1/users/me" \
  -H "Content-Type: application/json" \
  -H "Authorization: Bearer $(cat access_token.txt)"
```

The `jwt decode` command outputs my access token's claims:

```json
{
  "amr": [
    "pwd",
    "mfa",
    "otp"
  ],
  "aud": [
    "https://fleet-api.prd.na.vn.cloud.tesla.com",
    "https://fleet-api.prd.eu.vn.cloud.tesla.com",
    "https://auth.tesla.com/oauth2/v3/userinfo"
  ],
  "auth_time": 1700345533,
  "azp": "42faaff2d33a-xxx",
  "exp": 1700374340,
  "iat": 1700345540,
  "iss": "https://auth.tesla.com/oauth2/v3/nts",
  "locale": "en-AU",
  "ou_code": "NA",
  "scp": [
    "openid",
    "vehicle_device_data",
    "vehicle_cmds",
    "vehicle_charging_cmds"
  ],
  "sub": "6c81818d-xxx"
}
```

I asked for scopes: `openid user_data offline_access vehicle_device_data vehicle_cmds vehicle_charging_cmds`
But token only includes: `openid vehicle_device_data vehicle_cmds vehicle_charging_cmds`

And so I cannot call `/api/1/users/me` or similar `user_data` endpoints.

Anyone seen this and have a solution?

drnic, 2023-11-19:
Also odd: when logging into developer.tesla.com we do the oauth2 dance, but they use an older auth endpoint https://auth.tesla.com/en_au/oauth2/v1/authorize instead of https://auth.tesla.com/oauth2/v3/authorize which we're asked to use.

Interestingly they pass through a request for some different scopes: `user profile ou_code email`

I tried these on v3/authorize and `user` is dropped/not included in the access token, but `profile ou_code email` are present. Sadly, I still cannot call `/api/1/users/me` or other `user_data` endpoints

drnic, 2023-11-23:
@ThomasAlxDmy @sethterashima @fabianhu Can you help spot or suggest why `user_data` scope might not be appearing in refresh token nor access token?

Urkman, 2023-12-07:
I tried every endpoint that needs the `user_data ` scope. None of them are working.
Every endpoint returns `Unauthorized missing scopes`

@sethterashima can you help us with this? Or name someone who can help.
I already wrote an email to the support over one week ago, but no answer.

fanhaipeng0403, 2023-12-08:
However, accessing the API endpoint "/api/1/users/orders" is still not possible, and the documentation does not specify which authorization scope it belongs to. The error persists.

Urkman, 2023-12-08:
@fanhaipeng0403 
Where did you found information about adding " profile email " to the scopes?

drnic, 2023-12-08:
I was not able/allowed to authorized via the `.cn` audience `https://fleet-api.prd.cn.vn.cloud.tesla.com`; and as above the `.na` audience route does allow me to authorize but we are missing the `user_data` scope in our tokens :/

Urkman, 2023-12-09:
I tried with `.eu` audience.

Urkman, 2023-12-11:
@sethterashima This is still not working :(
Can you pleas help us, as we need to update our apps?

hardont943, 2023-12-12:
> @sethterashima This is still not working :( Can you pleas help us, as we need to update our apps?

估计中国地区的API还没开发好吧，整个流程还调不通

Urkman, 2023-12-18:
I nobody interested in fixing this?

NikhilTank35, 2023-12-26:
I passed openid%20offline_access%20user_data%20vehicle_device_data%20vehicle_cmds%20vehicle_charging_cmds in scope but it only takes openid, offline_access, and vehicle_device_data.

@ThomasAlxDmy @sethterashima @fabianhu Could you please help me with this?

[image]

aaronpkahn, 2024-04-24:
FYI all, we now support user scope updates from within accounts.tesla.com.  This is documented on #5 in the https://developer.tesla.com/docs/fleet-api#third-party-token authentication section.  Applications can direct users to https://auth.tesla.com/user/revoke/consent?revoke_client_id=$CLIENT_ID&back_url=$RETURN_URL to request additional scopes or revocation.

thefireblade, 2025-01-22:
Closing old issue

closed 2025-01-22 by thefireblade