teslamotors/vehicle-command #51: Waking up vehicle via BLE fails despite private key

top-gun, 2023-11-19:
I am trying to remote-control my vehicle via BT to implement PV surplus charging. Everything is fine via Homeassistant and the old legacy API, but I want to migrate to the new API.

I have set up a working key-pair, and I can unlock/lock etc my car from the command line. For example, locking the car works just fine with the following command:

> tesla-control -key-file private.pem -vin LRW3E7FS9PC82xxxx -ble lock

The car will lock up within seconds.

However, waking my vehicle up won't work. I would expect this command to wake up my vehicle:

tesla-control -key-file private.pem -vin LRW3E7FS9PC82yyyy -ble wake

However, I get the following response:

pi@teslamate:~ $ tesla-control -key-file private.pem -vin LRW3E7FS9PC821796 -ble wake
You must provide a private key with -key-name or -key-file to execute this command

The key exists in the current directory, I just used it seconds ago to lock the car. It works with all the commands I tried, only "wake" tells my I have not provided a key-file.

This is the debug output when I use wake:

---
pi@teslamate:~ $ tesla-control -debug -key-file private.pem -vin LRW3E7FS9PC82xxxx -ble wake
2023-11-19T23:56:38+01:00 [debug] Set session cache file to ''
2023-11-19T23:56:38+01:00 [debug] Set OAuth token name to ''
2023-11-19T23:56:38+01:00 [debug] Set OAuth token file to ''
2023-11-19T23:56:38+01:00 [debug] Set keyring type to ''
2023-11-19T23:56:38+01:00 [debug] Set keyring Debug Logging to 'false'
2023-11-19T23:56:38+01:00 [debug] Skipping private key loading because FlagPrivateKey is not set
2023-11-19T23:56:38+01:00 [debug] No private key available
2023-11-19T23:56:38+01:00 [debug] Connecting over BLE...
2023-11-19T23:56:38+01:00 [debug] Creating new BLE device
2023-11-19T23:56:38+01:00 [debug] Searching for BLE beacon S9f978dc8a8f1656dC...
2023-11-19T23:56:38+01:00 [debug] Connecting to BLE beacon...
2023-11-19T23:56:40+01:00 [info ] Connected to vehicle BLE
2023-11-19T23:56:40+01:00 [info ] Connecting to car...
2023-11-19T23:56:40+01:00 [info ] Starting dispatcher service...
2023-11-19T23:56:40+01:00 [warn ] No session available for DOMAIN_VEHICLE_SECURITY
You must provide a private key with -key-name or -key-file to execute this command
pi@teslamate:~ $

---

haexpo, 2023-11-20:
Same problem/error here.

top-gun, 2023-11-20:
I solved this by looking at the commands.go file and setting the parameter

		requiresAuth:     true,

instead of false. Another go build ./... and go install ./... and the command accepted my private key. 

I submitted https://github.com/teslamotors/vehicle-command/pull/54 for this issue.

haexpo, 2023-11-26:
I still can't wake the car. It always ends in "Error: context deadline exceeded":

pi@raspiBLE:~ $ sudo ./go/bin/tesla-control -vin 5YJXXXXXXXXXXX -ble -debug -key-file go/bin/private.pem wake
2023-11-25T14:06:42+01:00 [debug] Set session cache file to ''
2023-11-25T14:06:42+01:00 [debug] Set OAuth token name to ''
2023-11-25T14:06:42+01:00 [debug] Set OAuth token file to ''
2023-11-25T14:06:42+01:00 [debug] Set keyring type to ''
2023-11-25T14:06:42+01:00 [debug] Set keyring Debug Logging to 'false'
2023-11-25T14:06:42+01:00 [debug] Client public key: 04eff65c3fdb6702cd50926b60d9fb2639780b1XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
2023-11-25T14:06:42+01:00 [debug] Connecting over BLE...
2023-11-25T14:06:42+01:00 [debug] Creating new BLE device
2023-11-25T14:06:42+01:00 [debug] Searching for BLE beacon Se63aaba403c2ce80C...
2023-11-25T14:06:42+01:00 [debug] Connecting to BLE beacon...
2023-11-25T14:06:45+01:00 [info ] Connected to vehicle BLE
2023-11-25T14:06:45+01:00 [info ] Connecting to car...
2023-11-25T14:06:45+01:00 [info ] Starting dispatcher service...
2023-11-25T14:06:45+01:00 [info ] Securing connction...
2023-11-25T14:06:45+01:00 [info ] Requesting session info from DOMAIN_INFOTAINMENT
2023-11-25T14:06:45+01:00 [info ] Requesting session info from DOMAIN_VEHICLE_SECURITY
2023-11-25T14:06:45+01:00 [debug] TX: 320208023a1212105164b0edfa4f7502d02ffeeee7b672929a0310fb0003d453e9f246193c017bd005f64172430a4104eff65c3fdb6702cd50926b60d9fb2639780b1e8913db15d8351e51101d6809470717c135f0ba4f89b8341c82ae25b02d3197b20d23bb7a74172e0df5314532b7
2023-11-25T14:06:45+01:00 [debug] TX: 320208033a1212108172d50e140fc987587eca0b247e70899a0310476cfe94f7dd702917c23550d3eea73a72430a4104eff65c3fdb6702cd50926b60d9fb2639780b1e8913db15d8351e51101d6809470717c135f0ba4f89b8341c82ae25b02d3197b20d23bb7a74172e0df5314532b7
2023-11-25T14:06:46+01:00 [debug] RX: 321212105164b0edfa4f7502d02ffeeee7b672923a0208027a5a124104b66ff53613de4356282c93d277d14f9b3a5b81f3732c4b441939f1e8585185b0624b5ae70dc0bbac196bf7c5b577d86d0f614ab1fd04424fe40276ea27f20db51a109270075a8750712c6899f5482e34d71225fa0100006a2432220a2056107a6aeaa359f654f02c0f7ff588d008eba844e1bcfea6b8b3dca7d3ad2d3f920310fb0003d453e9f246193c017bd005f641
2023-11-25T14:06:46+01:00 [info ] [fb0003d453e9f246193c017bd005f641] Updated session info for DOMAIN_VEHICLE_SECURITY
2023-11-25T14:06:46+01:00 [debug] RX: 320208003a02080252071a051802220101
2023-11-25T14:06:46+01:00 [debug] [00] Dropping message to DOMAIN_BROADCAST
2023-11-25T14:06:47+01:00 [debug] RX: 320208003a02080252071a051802220101
2023-11-25T14:06:47+01:00 [debug] [00] Dropping message to DOMAIN_BROADCAST
2023-11-25T14:06:48+01:00 [debug] RX: 320208003a02080252071a051802220101
2023-11-25T14:06:48+01:00 [debug] [00] Dropping message to DOMAIN_BROADCAST
2023-11-25T14:06:49+01:00 [debug] RX: 320208003a02080252071a051802220101
2023-11-25T14:06:49+01:00 [debug] [00] Dropping message to DOMAIN_BROADCAST
2023-11-25T14:06:50+01:00 [debug] RX: 320208003a02080252071a051802220101
2023-11-25T14:06:50+01:00 [debug] [00] Dropping message to DOMAIN_BROADCAST
2023-11-25T14:06:51+01:00 [debug] RX: 320208003a02080252071a051802220101
2023-11-25T14:06:51+01:00 [debug] [00] Dropping message to DOMAIN_BROADCAST
2023-11-25T14:06:52+01:00 [debug] RX: 320208003a02080252071a051802220101
2023-11-25T14:06:52+01:00 [debug] [00] Dropping message to DOMAIN_BROADCAST
Error: context deadline exceeded

top-gun, 2023-11-26:
You need to restrict the domain to vcsec, or the call won't work.

This works for me. 

tesla-control **-domain vcsec** -ble -key-file private.pem -vin 1234xyzabc wake

haexpo, 2023-11-26:
Great, thank you! That solved the problem!

fanlessfan, 2023-12-12:
Hello @top-gun, Could you sharing the process on how to pair the key? Has vehicle have to be in specific state? I got error "failed to find BLE beacon for LRWYGDEEOPC" when try to run "tesla-control -debug -ble add-key-request"

closed 2024-03-19 by sethterashima

pkuehnel, 2024-07-10:
@top-gun How did you know that `-domain vcsec` is needed. Are there any other commands where this (or another) domain is needed?

top-gun, 2024-07-10:
It was a hint from a fellow on TFF ;)