teslamotors/vehicle-command #52: Car status information

haexpo, 2023-11-20:
I have successfully installed the BLE-Version and can send commands to both my Tesla.

Is it possible to get some status information from the car (e. g. "charging", "sleeping", ...) over the BLE-Interface?

top-gun, 2023-11-24:
This is a must for many usecases.

I use the API to charge when I have PV surplus or when the grid power is cheap. But to control this task, I need to know the battery SOC and whether the cable is plugged. It would be bad to send commands otherwise, as it would keep the car awake and waste battery energy,

haexpo, 2023-11-24:
Total blind flight without "vehicle_data"!

top-gun, 2023-11-24:
> Total blind flight without "vehicle_data"!

Exactly. Atm, I get vehicle data in Home Assistant via the Tesla integration, but that's based on the old API and non-commercial. Having vehicle data via BLE - even if it's only basic stuff like SOC, charge port and wake/sleep - would allow me to completely migrate to the BLE side of the new API.

BogdanDIA, 2024-01-01:
Hi, I have the same issue. How can I implement a proper automation with BLE commands if no status can be read from the car?

Any news on this ?

i1skn, 2024-02-19:
I also think, that adding "vehicle_data" is a huge must have. Without it it's impossible to create BLE only applications, as you still need vehicle data to be loaded from the web.

erikgieseler, 2024-05-28:
Is there any news on this?

horvatbenjamin, 2024-06-10:
Any news? It would be great!

m-chandler, 2024-06-20:
+1 this would awesome. Also using this for solar charging automation (as the Fleet API is a pain, and rate limited). Ideally I'd like to do this all with BLE.

At the moment I'm just manually turning off my solar charging automation when the car is plugged in / overnight, otherwise I'd be sending endless commands (keeping the car awake) as it's impossible to know the vehicle state via BLE.

thefireblade, 2025-01-22:
Support was added in https://github.com/teslamotors/vehicle-command/pull/330, closing this issue

closed 2025-01-22 by thefireblade