teslamotors/vehicle-command #53: Unable to Share Vehicle - Problem with enrolling Public Key

danielgUpside, 2023-11-20:
I'm attempting to enroll a public key onto a UK based tesla. However, when I go through to `tesla.com/_ak/<your-domain>` and select finish setup, it redirects to the tesla app and shows the following error. 

[image]

I've managed to register the public key using the `POST /api/1/partner_accounts` endpoint and its successfully ran. However I also then called the `GET /api/1/partner_accounts/public_key` endpoint which returns the following error message:

`user out of region, use base URL: https://fleet-api.prd.na.vn.cloud.tesla.com, see https://developer.tesla.com/docs/fleet-api#regional-requirements`.

I wonder if there is issues with registering public keys on EU/UK based vehicles. Has anyone else had this issue?

L480, 2023-11-27:
Had the same issue after only posting the registration against the North American endpoint. Make sure to [register](https://developer.tesla.com/docs/fleet-api#register) against `https://fleet-api.prd.eu.vn.cloud.tesla.com/api/1/partner_accounts` instead of `https://fleet-api.prd.na.vn.cloud.tesla.com/api/1/partner_accounts` if your vehicle is located in EMEA. The weird thing is that you need to do all following calls against the North American endpoint because otherwise you will still get the "out of region" error.

sethterashima, 2024-03-08:
Trying to close out old issues. There have been a few changes to the mobile app since this Issue was created. Are you still encountering it? If not, please close. Otherwise I can forward to the mobile app team.

itsMeDavidV, 2024-03-08:
@sethterashima "Finish Setup" button in Virtual Key webpage still does not work. It just reloads the webpage.

Also on iOS if the user does not have Safari set as their default browser the Tesla app won't open when VK link is opened.

itsMeDavidV, 2024-03-08:
> @sethterashima "Finish Setup" button in Virtual Key webpage still does not work. It just reloads the webpage.

Video:
https://github.com/teslamotors/vehicle-command/assets/12491192/d1f8c893-6190-47ec-ae5a-95c8d82468ac

Bu-TianLie, 2024-06-05:
tesla.com/_ak/<your-domain> ,Your domain name must be complete, e.g. www.example.com; I didn't fill in the "www" the first time I tested, and then couldn't share the vehicle, indicating that the program failed to register.

Bu-TianLie, 2024-06-05:
tesla.com/_ak/<your-domain>  must be complete, e.g. https://www.tesla.cn/_ak/www.example.com; I didn't fill in the "www" the first time I tested, and then couldn't share the vehicle, indicating that the program failed to register.
You also need to pay attention to what "https://www.tesla.cn" is in your area. In my case it's "https://www.tesla.cn", but in most areas it's https://www.tesla.com.

thefireblade, 2025-01-22:
Closing out old issue

closed 2025-01-22 by thefireblade