teslamotors/vehicle-command #56: Proxy: `actuate_trunk` always sends "vehicle not awake" with error 503

rawmean, 2023-11-22:
When sending an `actuate_trunk` command, the proxy does send the signed command to the car and the car does act on it correctly, but the proxy returns a 503 http error. This happens when the car is actually awake.

`2023-11-22T19:57:04Z [debug] Server returned 503: Service Unavailable: {"response":null,"error":"vehicle is not available","error_description":""}`

stx, 2023-11-22:
Same problem.

**This is a safety issue because the proxy retries the "failed" command automatically. This means it opens and closes the trunk repeatedly.**

As an emergency measure we have disabled trunk controls for Vehicle Command Protocol vehicles.

rawmean, 2023-11-27:
My workaround is to change the proxy code to send a different error and then process that error in a special way on the client. Couldn't figure out how to fix the proxy code properly

itsMeDavidV, 2023-12-06:
For me actuate_trunk always return vehicle is asleep error. The command never sends. Other command work fine. Has anyone fixed this issue?

stx, 2023-12-08:
@sethterashima Still happening. I'm testing on a 2023 Model X but this is happening to many different vehicles across the Tesla fleet. Email me at james@tessie.com and I can provide more details to help you reproduce.

Ran this just now against signed_command and this is actually moving the trunk:

```
./tesla-control trunk-move
Failed to execute command: vehicle unavailable: vehicle is offline or asleep
```

Urkman, 2023-12-12:
I also tried this an I get a 500 back...

Here is the log:

> 2023-12-12T16:48:45Z [info ] Received POST request for /api/1/vehicles/vin/command/flash_lights
2023-12-12T16:48:45Z [debug] Executing flash_lights on vin
2023-12-12T16:48:45Z [info ] Starting dispatcher service...
2023-12-12T16:48:45Z [info ] Requesting session info from DOMAIN_INFOTAINMENT
2023-12-12T16:48:45Z [debug] Sending request to https://fleet-api.prd.eu.vn.cloud.tesla.com/api/1/vehicles/vin/signed_command: {"routable_message":"MgIIAzoSEhBWn23cx3h4xGbytAL4jR18mgMQvcfV69XRxhUPbDxiTqMNfnJDCkEEe+Cj20FNv4ixztUa4jFKQZIMgAmS3+h43NESWAjg8jsAKuUed6wH5KhJ6UeSy7hSTvYEEDqX4tds5sLnmBpJlQ=="}
2023-12-12T16:48:45Z [info ] Requesting session info from DOMAIN_VEHICLE_SECURITY
2023-12-12T16:48:45Z [debug] Sending request to https://fleet-api.prd.eu.vn.cloud.tesla.com/api/1/vehicles/vin/signed_command: {"routable_message":"MgIIAjoSEhCW1RhUajMsr56dyssWwMYKmgMQslEi93wmS/JO/73yibhXanJDCkEEe+Cj20FNv4ixztUa4jFKQZIMgAmS3+h43NESWAjg8jsAKuUed6wH5KhJ6UeSy7hSTvYEEDqX4tds5sLnmBpJlQ=="}
2023-12-12T16:48:47Z [debug] Server returned 408: Request Timeout: {"response":null,"error":"vehicle is offline","error_description":""}
2023-12-12T16:48:47Z [warn ] [bdc7d5ebd5d1c6150f6c3c624ea30d7e] Terminal transmission error: vehicle unavailable: vehicle is offline or asleep
2023-12-12T16:48:47Z [error] Returning error Internal Server Error
2023-12-12T16:48:47Z [debug] [b25122f77c264bf24effbdf289b8576a] Retrying transmission after error: Post "https://fleet-api.prd.eu.vn.cloud.tesla.com/api/1/vehicles/vin/signed_command": context canceled

sethterashima, 2023-12-13:
@stx Can you run with `-debug` and post output?

stx, 2023-12-13:
@sethterashima 

Here you go. Email me if you need the redacted signed messages.

```
./tesla-control -debug trunk-move
2023-12-13T13:09:42-08:00 [debug] Set session cache file to ''
2023-12-13T13:09:42-08:00 [debug] Set keyring type to ''
2023-12-13T13:09:42-08:00 [debug] Set keyring Debug Logging to 'false'
2023-12-13T13:09:42-08:00 [debug] Client public key: 04c1e623abab4cf77a38f7d5dbfd2743f88b3d4661ec9131afa1b318fe8ee4ca77c9a74b096743f4407d2e69e6fcbc145836389fbed326b0f4c044910e769a5647
2023-12-13T13:09:42-08:00 [debug] Required OAuth parameters supplied by CLI and/or environment. Connecting over the Internet...
2023-12-13T13:09:42-08:00 [info ] Connecting to car...
2023-12-13T13:09:42-08:00 [info ] Starting dispatcher service...
2023-12-13T13:09:42-08:00 [info ] Securing connction...
2023-12-13T13:09:42-08:00 [info ] Requesting session info from DOMAIN_INFOTAINMENT
2023-12-13T13:09:42-08:00 [info ] Requesting session info from DOMAIN_VEHICLE_SECURITY
2023-12-13T13:09:42-08:00 [debug] Sending request to https://fleet-api.prd.na.vn.cloud.tesla.com/api/1/vehicles/REDACTED/signed_command: {"routable_message":"REDACTED"}
2023-12-13T13:09:42-08:00 [debug] Sending request to https://fleet-api.prd.na.vn.cloud.tesla.com/api/1/vehicles/REDACTED/signed_command: {"routable_message":"REDACTED"}
2023-12-13T13:09:43-08:00 [debug] Server returned 200: OK: {"response":"MhISEDvhXQfQp2MEKw2zLsshmi46AggDkgMQ4KjKHi8COXabOa9aVBa1GHpdCMgFEkEEE3Ft1rRl3d7dqVpDiTxnSuDG29G1PFAJc73IstGJhDdBk4hnjGGvFizhCY+olvLT1vYtfWlWJkZypkbIc0jlRRoQQJ6oNcqqaYyEmOzOyGU7oSU4pwMAaiQyIgogFGhNp3yS2QObwcc1yXNrIQ+IyIDcmgHyaR492xHvI0I="}
2023-12-13T13:09:43-08:00 [info ] [e0a8ca1e2f0239769b39af5a5416b518] Updated session info for DOMAIN_INFOTAINMENT
2023-12-13T13:09:44-08:00 [debug] Server returned 200: OK: {"response":"MhISEIczSxHQ/c6Ax8zkUszMhBM6AggCeloSQQT2gy/5fXUnDnOW2oIGJlioNfkDnqQovUh8S4vGECUI/JKkZJ8rADVz7bH7iAGz9xz71ZoyLnDZ6bBGPti+2jNVGhA/4SR1jnfMfVPxvXsnGlPtJQmtAwBqJDIiCiAFiWaa0cOC8xb6EzO9B4amxTEq+6KQEg9HMcts3g6o65IDELjSf+wisEWtRgmCThq4F0Q="}
2023-12-13T13:09:44-08:00 [info ] [b8d27fec22b045ad4609824e1ab81744] Updated session info for DOMAIN_VEHICLE_SECURITY
2023-12-13T13:09:44-08:00 [debug] Sending request to https://fleet-api.prd.na.vn.cloud.tesla.com/api/1/vehicles/REDACTED/signed_command: {"routable_message":"REDACTED"}
2023-12-13T13:09:49-08:00 [debug] Server returned 503: Service Unavailable: {"response":null,"error":"vehicle is not available","error_description":""}
2023-12-13T13:09:49-08:00 [warn ] [4e27eb2bab1b3d6753e9fa541759e050] Terminal transmission error: vehicle unavailable: vehicle is offline or asleep
Failed to execute command: vehicle unavailable: vehicle is offline or asleep
```

stx, 2023-12-13:
@sethterashima 

Here you go. Email me if you need the redacted info.

```
./tesla-control -debug trunk-move
2023-12-13T13:09:42-08:00 [debug] Set session cache file to ''
2023-12-13T13:09:42-08:00 [debug] Set keyring type to ''
2023-12-13T13:09:42-08:00 [debug] Set keyring Debug Logging to 'false'
2023-12-13T13:09:42-08:00 [debug] Client public key: REDACTED
2023-12-13T13:09:42-08:00 [debug] Required OAuth parameters supplied by CLI and/or environment. Connecting over the Internet...
2023-12-13T13:09:42-08:00 [info ] Connecting to car...
2023-12-13T13:09:42-08:00 [info ] Starting dispatcher service...
2023-12-13T13:09:42-08:00 [info ] Securing connction...
2023-12-13T13:09:42-08:00 [info ] Requesting session info from DOMAIN_INFOTAINMENT
2023-12-13T13:09:42-08:00 [info ] Requesting session info from DOMAIN_VEHICLE_SECURITY
2023-12-13T13:09:42-08:00 [debug] Sending request to https://fleet-api.prd.na.vn.cloud.tesla.com/api/1/vehicles/REDACTED/signed_command: {"routable_message":"REDACTED"}
2023-12-13T13:09:42-08:00 [debug] Sending request to https://fleet-api.prd.na.vn.cloud.tesla.com/api/1/vehicles/REDACTED/signed_command: {"routable_message":"REDACTED"}
2023-12-13T13:09:43-08:00 [debug] Server returned 200: OK: {"response":"MhISEDvhXQfQp2MEKw2zLsshmi46AggDkgMQ4KjKHi8COXabOa9aVBa1GHpdCMgFEkEEE3Ft1rRl3d7dqVpDiTxnSuDG29G1PFAJc73IstGJhDdBk4hnjGGvFizhCY+olvLT1vYtfWlWJkZypkbIc0jlRRoQQJ6oNcqqaYyEmOzOyGU7oSU4pwMAaiQyIgogFGhNp3yS2QObwcc1yXNrIQ+IyIDcmgHyaR492xHvI0I="}
2023-12-13T13:09:43-08:00 [info ] [e0a8ca1e2f0239769b39af5a5416b518] Updated session info for DOMAIN_INFOTAINMENT
2023-12-13T13:09:44-08:00 [debug] Server returned 200: OK: {"response":"MhISEIczSxHQ/c6Ax8zkUszMhBM6AggCeloSQQT2gy/5fXUnDnOW2oIGJlioNfkDnqQovUh8S4vGECUI/JKkZJ8rADVz7bH7iAGz9xz71ZoyLnDZ6bBGPti+2jNVGhA/4SR1jnfMfVPxvXsnGlPtJQmtAwBqJDIiCiAFiWaa0cOC8xb6EzO9B4amxTEq+6KQEg9HMcts3g6o65IDELjSf+wisEWtRgmCThq4F0Q="}
2023-12-13T13:09:44-08:00 [info ] [b8d27fec22b045ad4609824e1ab81744] Updated session info for DOMAIN_VEHICLE_SECURITY
2023-12-13T13:09:44-08:00 [debug] Sending request to https://fleet-api.prd.na.vn.cloud.tesla.com/api/1/vehicles/REDACTED/signed_command: {"routable_message":"REDACTED"}
2023-12-13T13:09:49-08:00 [debug] Server returned 503: Service Unavailable: {"response":null,"error":"vehicle is not available","error_description":""}
2023-12-13T13:09:49-08:00 [warn ] [4e27eb2bab1b3d6753e9fa541759e050] Terminal transmission error: vehicle unavailable: vehicle is offline or asleep
Failed to execute command: vehicle unavailable: vehicle is offline or asleep
```

sethterashima, 2023-12-15:
@stx : Looks like there might be two related issues: (1) the SDK isn't retrying correctly on a 503 and (2) The 503 should not be easily reproducible when the vehicle is online, as seen by the successful handshake messages that return 200. The fleetnet team is investigating the second issue and may reach out to you separately.

As for (1), could you try the code from this branch to see if it makes actuate_trunk more reliable: https://github.com/teslamotors/vehicle-command/pull/108?

rawmean, 2023-12-17:
> @stx : Looks like there might be two related issues: (1) the SDK isn't retrying correctly on a 503 and (2) The 503 should not be easily reproducible when the vehicle is online, as seen by the [...]

I can reproduce the problem every single time the "open trunk" message is sent to proxy. Specifically, I make sure while is awake and then send the "open trunk" message. It opens the trunk but then it issues a 503 error.

itsMeDavidV, 2023-12-18:
So I just reproduced this issue while not using the SDK, same exact behavior/result:

trunk opens, 503 returned. Each manual retry produces the same result. The only way to confirm is via checking vehicle_state.

I also noticed that (at least for trunk) `CLOSURE_MOVE_TYPE_OPEN` & `CLOSURE_MOVE_TYPE_CLOSE` flat out just do not work. The vehicle returns 503 and the trunk does not move at all.

rawmean, 2023-12-19:
> @sethterashima so I just reproduced this issue while not using the SDK, same exact behavior/result:

What do you mean by "while not using the SDK". How else are you sending the command to the car?

itsMeDavidV, 2023-12-20:
> How else are you sending the command to the car?

by sending a POST request directly to [/api/1/vehicles/{vehicle_tag}/signed_command](https://developer.tesla.com/docs/fleet-api#share_invites-revoke:~:text=signed_command,-POST%20/api/1)

rawmean, 2023-12-20:
> > How else are you sending the command to the car?
> 
> by sending a POST request directly to [/api/1/vehicles/{vehicle_tag}/signed_command](https://developer.tesla.com/docs/fleet-api#share_invites-revoke:~:text=signed_command,-POST%20/api/1)

I see. By signing the message yourself using the recently published protos?

Urkman, 2023-12-31:
This is still not fixed, correct?

avinoamr, 2024-02-28:
Is this still an issue?

itsMeDavidV, 2024-02-28:
I can check when I get back to my Mac, but this was a vehicle sever issue not an issue with the proxy

stx, 2024-02-28:
@avinoamr Yep, this looks fixed now. Thank you!

closed 2024-06-25 by sethterashima