teslamotors/vehicle-command #57: rate limits are not currently applied

jbanyer, 2023-11-23:
The new API describes rate limits under the Discovery Plan.

However in my limited testing these limits don't seem to be applied currently. Specifically:
- one vehicle data per 5 minutes (I can send one call per minute)
- 50 commands per day per vehicle (I can send over 100 commands per day)

Is there a timeframe for implementing the rate limits?

jbanyer, 2023-12-11:
Any chance of an update here?

We'd like to prepare for the rate limits before they become effective, but it's difficult without knowing exactly how they work.

For example, regarding the "vehicle data" 1 call per 5 minutes per vehicle limit. Currently we poll vehicles like this:
- `GET /api/1/vehicles` - check if the vehicle is awake / online
- if awake: `GET /api/1/vehicles/${id_s}/vehicle_data`

If the first call counts toward the rate limit, the second call would always fail.

Do we need to change this?

ThomasAlxDmy, 2024-06-25:
We're currently giving free almost unlimited access. This will change in the next couple months.

vbarrier, 2024-06-25:
Hi, is this issue #255 cancelled?

closed 2024-06-25 by sethterashima