teslamotors/vehicle-command #58: Automatically select the desired vehicle on pairing

vbarrier, 2023-11-26:
Hi 

Would it be possible to add an anchor like #VIN_NUMBER (or vehicle id) at the end of https://tesla.com/_ak/example.com to automatically select the desired vehicle and do the pairing? 

Regards;

jbanyer, 2023-11-28:
Plus one for this. The current workflow is not very user friendly, eg:
- in our app, show instructions, something like:
- if you have multiple vehicles, switch to the Tesla app, select the correct vehicle
- switch back to this app
- click the pairing link, which will open the Tesla app
- accept the pairing request

corsair, 2023-12-03:
**+1 to this.** The current required flow results in a very frustrating UX for vehicle owners with multiple vehicles, and also adds complexity (as described well by @jbanyer).

I have had multiple vehicle owners with 6-12 vehicles express their displeasure, which would be mostly resolved by defining a vehicle ID or VIN to the URL to direct the Tesla App to the desired vehicle.

vbarrier, 2024-06-14:
Hi, any progress/workaround for this?

thefireblade, 2025-01-22:
This would be a request for the mobile app team, at the moment, we do not have plans on implementing this, but I understand that this is frustrating from a user experience perspective, I've filed an internal ticket and am closing this old issue

closed 2025-01-22 by thefireblade