teslamotors/vehicle-command #64: How to revoke access to the app

rawmean, 2023-11-28:
Is it possible to give the user a link so that they can revoke access to the app from within the app.
I know that the user can log in to their Tesla account and go to Account Settings / Security and manage access to 3rd-party apps, but that is a little cumbersome.

The email that Tesla sends when the user gives permissions to the app contains a link to revoking access. But I am not sure if that link works for all users of the app or if it is specific to a particular user.

jbanyer, 2023-11-29:
Plus one for this, or something similar. We are finding that users are failing to select the required permissions when granting access, and this leaves a complicated situation to clean up.

An automated way to revoke access would help.

Alternatively, if we could require that the user selects all the required permissions, and reject the access grant if they don't, that would avoid the problem in the first place. Is there a way to do this?

jbanyer, 2023-11-29:
It appears you can give the user a direct link to revoke access `https://auth.tesla.com/en_au/user/revoke/consent?revoke_client_id=<your-app-id>`

This takes the user to the tesla.com page for revoking access, and they just click one button. (after signing in)

closed 2023-11-30 by rawmean