teslamotors/vehicle-command #67: Endpoint for China?

rawmean, 2023-11-29:
The documentation does not specify an endpoint for China. What endpoint are we supposed to use for China?
[image]

itsMeDavidV, 2023-12-08:
Any update on this?

ixiaogangwyg1, 2023-12-11:
Have been updated👇
China: [https://fleet-api.prd.cn.vn.cloud.tesla.cn](https://developer.tesla.com/fleet-api.prd.cn.vn.cloud.tesla.cn) - localized documentation at https://developer.tesla.cn/docs/fleet-api

rawmean, 2023-12-11:
I wonder if this is going to work automatically if we are using the `region` endpoint.
Testing this is challenging without having a car in China