teslamotors/vehicle-command #71: [Gomobile] unsupported parameter/ return types

itsMeDavidV, 2023-11-30:
When using Gomobile to create bindings there are a lot of methods left out due to unsupported parameter/ return types. Will you (Tesla) address this?

For example in pkg/account, the following methods can't be included when creating a binding for iOS:

```
Account.Get
Account.GetVehicle
Account.Post
Account.SendVehicleFleetAPICommand
Account.UpdateKey
```

sethterashima, 2023-12-02:
It's unlikely that we'll add support for Gomobile. You'll have to create a wrapper around vehicle-command to get that functionality. That being said, please feel free to suggest tweaks to the API if they would make this task significantly easier.

itsMeDavidV, 2023-12-02:
@sethterashima thanks for the heads up. In that case, would it be possible for mobile devs to have access to `.proto`s or mobile friendly `.pb`s (eg: `.pb.swift`)and [use /api/1/vehicles/{id}/signed_command directly?](https://github.com/teslamotors/vehicle-command/issues/72)

> That being said, please feel free to suggest tweaks to the API if they would make this task significantly easier.

Unfortunately, bc Gomobile's acceptable types are so restricted this might not be likely without a lot of changes. I'm not familiar with Golang enough to feel confident doing this, but might I suggest to the team (or someone else who wants to submit a PR) forking Gomobile and creating an alternate version that can accept the custom types found in vehicle-command. Aside from the alt option I mentioned above, I think this would be the best option if we are aiming to not make many changes while being able to easily onboard mobile devs.

sethterashima, 2023-12-06:
Duplicate of https://github.com/teslamotors/vehicle-command/issues/62. Proto files are coming soon.