teslamotors/vehicle-command #72: Can we use /api/1/vehicles/{id}/signed_command directly?

itsMeDavidV, 2023-11-30:
For a mobile use case, Go--while i'm sure is great in many cases--is not ideal in a native mobile setting. As an alternative can we use `/api/1/vehicles/{id}/signed_command` directly? Ideally also having access to the original protobufs instead of the pb.go versions?

jbanyer, 2023-11-30:
This is pretty much the same as #37. Fee free to close one of them :)

itsMeDavidV, 2023-11-30:
> This is pretty much the same as #37. Fee free to close one of them :)

Yes and I'm a little saddened yours hasn't been answered yet given the timeline on deprecation. But differentiating here, I'm not asking if its possible, it clearly is. I just want to know can we get the original protobufs and **is Tesla okay with us doing this instead of using the SDK**.

2 aims here are:

1: To only use non-edited Tesla source for SDK **OR** manually make network requests.
2: Not do anything Tesla wouldn't approve of (eg: going outside of the bounds of their intended use for their API)

> To only use non-edited Tesla source for SDK
Right now this can't be achieved without using an intermediary.

sethterashima, 2023-12-02:
You are free to hit that endpoint using another client. Please set the user agent to something descriptive.

We hope to release the proto files soon. Just need some cleanup and a more maintainable system for the managing open-source versions in parallel.

sethterashima, 2023-12-06:
> Where can we find information on how to "sign" the command?

Protos are now available and documentation is coming soon. But since that's outside the scope of this issue as clarified above, I'm closing it out.

closed 2023-12-06 by sethterashima

itsMeDavidV, 2023-12-07:
> > Where can we find information on how to "sign" the command?
> 
> Protos are now available and documentation is coming soon. But since that's outside the scope of this issue as clarified above, I'm closing it out.

Thank you tons! This will help the mobile community a lot :)

eiannone, 2023-12-31:
Hi, I made a very simple Node.js client which can send signed commands directly to the API.
The project is just a draft, with the purpose of understanding the basics of Tesla protobuf message signing. I've implemented the handshake part to start an authenticated session, and a CarServer class, which can send signed messages.
You can find the code here: https://github.com/eiannone/tesla-cmd-api

Urkman, 2024-01-05:
@eiannone That sounds great :) 
But I'm not a node developer, so the code ist hard to read.
Could you please write down some steps you did to get it working?
( Still waiting for some official docs :( )