teslamotors/vehicle-command #74: allow editing third party app scopes

jbanyer, 2023-12-01:
Currently the developer portal does not allow editing scopes once an app has been created.

This will be a problem if an app needs new scope access to support new features, but the scope was not selected on creation.

Is there any reason this scope list can't be edited?

It won't impact existing users and tokens since they've already selected their scopes. It only impacts new auth grants?

eclipsed4utoo, 2023-12-05:
Just to add, it would also be nice to update the URLs associated with the app in-case of a domain/app name change.

llamafilm, 2024-03-31:
As a workaround for this issue, I tried registering a new app, but when I try to register at `/api/1/partner_accounts` I get error 422: {"response":null,"error":"Validation failed: Domain has already been taken","error_description":""}.  So that requires changing domains too.

ThomasAlxDmy, 2024-06-25:
This is now available via the UI.