teslamotors/vehicle-command #76: "unsigned_cmds_hardlocked" message from new Fleet API

morganofslo, 2023-12-02:
Anyone else see this?  This was for a vehicle using the [new REST API](https://developer.tesla.com/docs/fleet-api#overview), not the old owner api.  I assume this means Tesla is feature flagging this vehicle to require signed commands via vehicle-command api ahead of the January launch?

`'status': 200, 'data': {'response': {'result': False, 'reason': 'unsigned_cmds_hardlocked'}}}`

Just odd that it's different than the other error they have documented:
`"Tesla Vehicle Command Protocol required, please refer to the documentation here: https://developer.tesla.com/docs/fleet-api#2023-10-09-rest-api-vehicle-commands-endpoint-deprecation-warning (403)"`

stx, 2023-12-02:
Yep.

rawmean, 2023-12-03:
It must be one of those cars that have not used the owner-api in the past 30 days. According to their notice, those cars will be forced to use the new API

closed 2023-12-21 by morganofslo

ChogosuJacob, 2024-02-10:
Any solution for this error?
I just got my new M3P and heard about Tasker.
So I purchased Tasker and Bolt for Tesla already and it's not working. :(