teslamotors/vehicle-command #83: HTTP Proxy - "Server stopped: x509: invalid ECDSA parameters"

eclipsed4utoo, 2023-12-06:
I am attempting to follow the directions for running the http proxy locally on my mac. I believe I followed the steps correctly, using the supplied `openssl` commands to create the localhost key and cert, and then providing those to the command to start the http proxy, but I am getting an error after the proxy starts...

"invalid ECDSA parameters"

(turning on verbose logging)
[image]

Any idea on what is going on? Did a google search for that error and it referenced possible issues with how Go processes x509 certificates. I'm assuming others have actually gotten the http proxy to work, so just looking for some pointers on how I can get mine to work.

rileymd88, 2023-12-06:
I ran into the same issue yesterday. You need to create a private key first:

https://github.com/teslamotors/vehicle-command#generating-a-command-authentication-private-key

eclipsed4utoo, 2023-12-06:
@rileymd88 what do I do with the key after it's generated? I already have one generated for my domain and put at the correct location of my domain, but when wanting to test this out locally, I'm not sure what to do with the key. 

and any idea why the section says "private key" when it creates a public key?

rileymd88, 2023-12-06:
I following the documentation and did this:

```
export TESLA_KEY_NAME=$(whoami)
tesla-keygen create > public_key.pem
```

rileymd88, 2023-12-06:
Just to add with this since I struggled with this tonight again. The following steps worked for me:

1. Delete the private key from Keychain Access found in the login Default Keychain. Not sure why this step is sometimes needed!
2. Run `tesla-keygen -f create > public_key.pem`
3. Run `tesla-http-proxy -tls-key key.pem -cert cert.pem -port 4443`

sethterashima, 2023-12-07:
@eclipsed4utoo Could you please paste:

- The certificate file (cert.pem)
- The command you use to start the server
- The output of `openssl ec -check -in key.pem -noout` (where `key.pem` is your TLS private key)

...and then confirm the output of `openssl asn1parse -in key.pem |grep -v OCTET` matches the text below:

```
    0:d=0  hl=3 l= 238 cons: SEQUENCE          
    3:d=1  hl=2 l=   1 prim: INTEGER           :00
    6:d=1  hl=2 l=  16 cons: SEQUENCE          
    8:d=2  hl=2 l=   7 prim: OBJECT            :id-ecPublicKey
   17:d=2  hl=2 l=   5 prim: OBJECT            :secp521r1
```

eclipsed4utoo, 2023-12-07:
> Just to add with this since I struggled with this tonight again. The following steps worked for me:
>
> 1. Delete the private key from Keychain Access found in the login Default Keychain. Not sure why [...]

I don't seem to have any private keys in Keychain Access that I can see that might relate to this. I have some for apple development, iMessage, and "exchange" keys, but none that seem related to what I'm doing here.

Should I be able to double-click/install any of these keys/certs into Keychain Access? or should they show up automatically?

Are you running on Apple hardware(M1/M2/M3)? or intel?

jbanyer, 2023-12-07:
For anyone struggling with Apple Keychain etc (I did too), you can just use plain files, eg:

```
bin/tesla-http-proxy -cert $CERT_PATH\
    -key-file $PRIVATE_KEY_PATH\
    -host 0.0.0.0 -port 443 -tls-key $TLS_KEY_PATH -verbose
```

sethterashima, 2023-12-07:
@eclipsed4utoo It looks like you're running into the Golang issue linked to in the above Stack Exchange answer: https://github.com/golang/go/issues/21502. (Your private key file use a standard name for the curve but instead gives the explicit set of parameters that define the curve).

What command did you use to generate the key file? You should use the OpenSSL command from the README (`openssl req ...`). If you are using that command, then it's possible you're using some unsupported version of OpenSSL; the fact that your OpenSSL version does not support `openssl ec -check ...` also points to this conclusion.

I'm not aware of any OpenSSL versions/implementations that behave this way. Could you run `openssl version`?

eclipsed4utoo, 2023-12-08:
> @eclipsed4utoo It looks like you're running into the Golang issue linked to in the above Stack Exchange answer: [golang/go#21502](https://github.com/golang/go/issues/21502). (Your private key file does not use a standard name for the curve but instead gives the explicit set of parameters that define the curve).
> 
> What command did you use to generate the key file? You should use the OpenSSL command from the README (`openssl req ...`). If you are using that command, then it's possible you're using some unsupported version of OpenSSL; the fact that your OpenSSL version does not support `openssl ec -check ...` also points to this conclusion.
> 
> I'm not aware of any OpenSSL versions/implementations that behave this way. Could you run `openssl version`?
> 
> Edit: As a work around, you could try converting from explicit to named:
> 
> ```
> cat key.pem |  openssl ec -param_enc named_curve > new_key.pem
> ```
> 
> ...and then try launching the proxy with `-tls-key new_key.pem`.

This was exactly my problem. Seems that LibreSSL is installed by default on Macs, which is a completely different version of openssl. So I installed homebrew and installed the latest OpenSSL version 1.1.1, updated the PATH so it would be used, and now the http-proxy seems to run correctly.

[image]

While it is an error, I can work through it...at least I am getting a response!!!!

Thanks to everybody for the help.

eclipsed4utoo, 2023-12-08:
I guess one more question, maybe for @sethterashima (or anybody else that could answer)

If I currently have an app that is running off of the old ownerapi, written in C#(so the golang SDK isn't really something I can use), I would need to:

- spin up my own server(assuming a VPS would be fine)
- start this http-proxy at localhost on that server
- update my API calls from `owner-api.teslamotors.com/api/1` to `[my-domain]/api/1`

Is that correct?

PR #95 (2023-12-08): README: Force use of named curves in example
MacOS uses LibreSSL for the OpenSSL command, and LibreSSL defaults to using explicit curve parameters rather than named curves when encoding private keys with `openssl req ...`. But the Golang x509 library does not support explicit curve parameters, causing `tesla-http-proxy` to fail with "x509: invalid ECDSA parameters".

This commit fixes the problem by adding an option to the openssl command in the README that forces LibreSSL to use named curves.

# Description

Please include a summary of the changes and the related issue.

Fixes #83 

## Type o [... cut]
closed 2023-12-08 by sethterashima