teslamotors/vehicle-command #89: Fleet API works but SDK/signed_command returning "vehicle is not available"

stx, 2023-12-07:
Fleet API commands work perfectly, but command protocol requests fail with "vehicle not awake" (the vehicle is awake and responsive.)

This appears to have randomly started happening this morning.

Via Fleet API: https://fleet-api.prd.na.vn.cloud.tesla.com/api/1/vehicles/[id]/command/door_lock

```
{"response":{"result":true,"reason":""}}
```

Via SDK proxy: https://localhost/api/1/vehicles/[vin]/command/door_lock

```
2023-12-07T20:25:58Z [info ] Received POST request for /api/1/vehicles/[vin]/command/door_lock
2023-12-07T20:25:58Z [debug] Executing door_lock on [vin]
2023-12-07T20:25:58Z [info ] Starting dispatcher service...
2023-12-07T20:25:58Z [info ] Requesting session info from DOMAIN_INFOTAINMENT
2023-12-07T20:25:58Z [info ] Requesting session info from DOMAIN_VEHICLE_SECURITY
2023-12-07T20:25:58Z [debug] Sending request to https://fleet-api.prd.na.vn.cloud.tesla.com/api/1/vehicles/[vin]/signed_command: {"routable_message":"..."}
2023-12-07T20:25:58Z [debug] Sending request to https://fleet-api.prd.na.vn.cloud.tesla.com/api/1/vehicles/[vin]/signed_command: {"routable_message":"..."}
2023-12-07T20:25:58Z [debug] Server returned 503: Service Unavailable: {"response":null,"error":"vehicle is not available","error_description":""}
2023-12-07T20:25:58Z [warn ] [af82c7e5bec4341f1444161cc755516f] Terminal transmission error: vehicle not awake
2023-12-07T20:25:58Z [debug] [ec6994d3f688d74d835489f7477a5f7e] Retrying transmission after error: Post "https://fleet-api.prd.na.vn.cloud.tesla.com/api/1/vehicles/[vin]/signed_command": context canceled
2023-12-07T20:25:58Z [error] Returning error Internal Server Error
```

itsMeDavidV, 2023-12-07:
Same happens for closure move requests for trunk and frunk in vehicle-command SDK. see [issue #56](https://github.com/teslamotors/vehicle-command/issues/56)

stx, 2023-12-07:
The underlying Fleet API behavior/issue is different between this and #56.

That other issue:
1. Has been going on for awhile.
2. Only affected actuate_trunk commands.
3. Vehicle actually executed the actuate_trunk command.

This issue is brand new, is affecting all commands and does not execute the command.

stx, 2023-12-07:
Definitely not limited to the proxy.

```
curl "https://fleet-api.prd.na.vn.cloud.tesla.com/api/1/vehicles/[id]/command/door_lock" -H "Authorization: Bearer [token]" -X POST
{"response":{"result":true,"reason":""}}

./tesla-control lock
Error: vehicle unavailable: vehicle is offline or asleep
```

stx, 2023-12-07:
@sethterashima It looks like signed_command was just fixed on Fleet API. Proxy and CLI working again. Thank you!

Looks like the signed_command issue on #56 remains though - "vehicle is not available" and it opens the trunk anyway.

itsMeDavidV, 2023-12-08:
> Looks like the signed_command issue on #56 remains though - "vehicle is not available" and it opens the trunk anyway.

@sethterashima yeah this is a big one. Any progress on addressing this?

thefireblade, 2025-01-22:
I believe we've dug into this in the past, going to close this old issue since not sure if this is causing problems right now. Feel free to reopen if this is still an issue

closed 2025-01-22 by thefireblade