teslamotors/vehicle-command #98: Receiving "MESSAGEFAULT_ERROR_INSUFFICIENT_PRIVILEGES" when waking up vehicle via BLE

L480, 2023-12-09:
I'm trying to wake up my Tesla via BLE as described in the [BLE example](https://github.com/teslamotors/vehicle-command/blob/main/examples/ble/main.go). Without any changes I end up with the same issue as described in https://github.com/teslamotors/vehicle-command/issues/51#issuecomment-1826769722, so I limited the domain to VCSEC. However, now I'm receiving a "MESSAGEFAULT_ERROR_INSUFFICIENT_PRIVILEGES":

```
2023-12-09T19:10:29+01:00 [debug] Creating new BLE device
2023-12-09T19:10:29+01:00 [debug] Searching for BLE beacon S1de62fd3767486a3d...
2023-12-09T19:10:29+01:00 [debug] Connecting to BLE beacon...
2023-12-09T19:10:30+01:00 [info ] Connected to vehicle BLE
2023-12-09T19:10:30+01:00 [info ] Starting dispatcher service...
2023-12-09T19:10:30+01:00 [info ] Requesting session info from DOMAIN_VEHICLE_SECURITY
2023-12-09T19:10:30+01:00 [debug] TX: 1210400ca1c7388bccad50697f69a0310b72a233c2802598239b5d530a4104f0ceb931f4d6b9b2381cf92093d2327ca2f488884cca5d775373e4ce7f791ef1b42f3c9e9baf1c8a9f6ad3a121097bf0259d6520
2023-12-09T19:10:31+01:00 [debug] RX: 1210400ca1c7388769cbff23bccad50697f63a0208027a5c080112410430b8b15691fdb513794e6ce79ae3ce8b7a91857ee6a328308a0a1ea79af04284c4cad0866caf14ad02e9859f9ea1a105bf6fa59f88888fbd64254b0100006a2432220a20be355c144f5a778cc6eeae20d716ae92e5edd59aa5d195f0c00920310ee85c16b72a233c280259
2023-12-09T19:10:31+01:00 [info ] [ee85c16b73c2802598239b0] Updated session info for DOMAIN_VEHICLE_SECURITY
2023-12-09T19:10:31+01:00 [debug] TX: 32021210a0b4bdf16a31c51f2f6eeb6a89a03109e188ecde1cb4608f160fe3ada4c798e510a430a4104f0ceb931f4d6b9b2dec8e4bc6381cf92093d2327ca2f488884cca5d775373e4cef324dcb7f791ef1b42f3c9e9baf1c8a9f6ad3a121097bf0259d6520be942fde2a390a105bf6f1c034711f9b7da59f88888fbd64120cd381bf10a1e022be8a1c4fc3180225500100002a1016208c1ca9689a50501ec58bc17810c5
2023-12-09T19:10:31+01:00 [debug] RX: 3202521f1a1d12160a146f9bc89b1131e7ec1b260e106538aa908bba561b1802220101
2023-12-09T19:10:31+01:00 [debug] [00] Dropping message to DOMAIN_BROADCAST
2023-12-09T19:10:31+01:00 [debug] RX: c51f2f63a0b4bdf16a31c51f2f6386f9afeeb6a83a0208025206220412020802
2023-12-09T19:10:32+01:00 [debug] RX: c51f2f636a31c51f2f6386f9afeeb6a83a0208027a5c080212410430b8c691fdb513794e6ce79ae3ce8b7a91857ee6a328308a0a1ea79af96d02a489c2e655af3004284c4cad0866ea1a105bf6f1c034711f9b7da59f88888fbd080210076a2432220a20e5759185a1c4bcb2fabc70a3509437ee4f4ceaca838cb1da9203109e188ecde1cb4
2023-12-09T19:10:32+01:00 [info ] [e1cb4608f160fe3ada4c798e] Updated session info for DOMAIN_VEHICLE_SECURITY
2023-12-09T19:10:32+01:00 [error] failed to start session: failed to wake up vehicle: MESSAGEFAULT_ERROR_INSUFFICIENT_PRIVILEGES
```

Any idea how to fix this?

L480, 2023-12-10:
It appears that adding the public key via `https://tesla.com/_ak/<your-domain>` is not the same as adding it through tesla-control over Bluetooth, even though it looks the same in the vehicle (both were displayed as "Flottenschlüssel"). It also seems that an "app public key" cannot coexist with a public key added via BLE. At least I had to replace the key coming from the app with the same key over Bluetooth for it to work. Can someone elaborate on that?

fanlessfan, 2023-12-12:
Hello @L480, Could you share the process on how to pair the key? I got error "failed to find BLE beacon for LRWYGDEEOPC".

thx

L480, 2023-12-12:
Sounds like you are out of BLE range of your vehicle.

fanlessfan, 2023-12-12:
I did this inside the vehicle. Should I go out of it?

sethterashima, 2023-12-15:
> It appears that adding the public key via https://tesla.com/_ak/<your-domain> is not the same as adding it through tesla-control over Bluetooth, even though it looks the same in the vehicle

This is correct, and we need to document the difference clearly.

If you add a key using the `https://tesla.com/_ak/<your-domain> ` flow, it gets enrolled as a "Fleet Manager" key. Starting in 2023.38, Fleet Manager keys do not work over BLE and cannot add/remove other keys. The rationale here is that if a big 3rd-party fleet manager private key is compromised, then Tesla can just revoke OAuth tokens and then the key can't be used for anything. No urgent need to worry about cleaning up keys from individual vehicles; they can't be stolen over BLE even with a compromised private key.

On the other hand, if you use `tesla-control` to add a key, it can be an Owner or Driver, both of which work over BLE. 

If you run `tesla-control -ble -vin xxx list-keys` you can see all the paired public keys and their assigned roles.

L480, 2023-12-15:
Makes sense, thanks for the clarification @sethterashima.

closed 2023-12-15 by L480