teslamotors/vehicle-command #99: `tesla.com/_ak/<your-domain>` opens Tesla app, but no pop up to approve/deny

rawmean, 2023-12-09:
Some users report that `tesla.com/_ak/<your-domain>` opens Tesla app, but no pop up to approve/deny. Any idea what the problem could be?

stx, 2023-12-11:
It's a bug in the Tesla app. The only solution is to force close the app.

fanhaipeng0403, 2023-12-29:
After my testing, the file "publickey.pem" under the "/.well-known" directory must be stored and made available for download by users, rather than being returned as HTML. This will significantly increase the pop-up rate after redirection.

good 
[image]

bad
[image]