teslamotors/vehicle-command #101: Error: failed to find BLE beacon for LRWYGDEEOPCxxxxxx

fanlessfan, 2023-12-12:
Hello,

I created the key pairs by tesla-keygen and try to add it to car and got error below. it looks like the private key was not loaded, the key file was set. I run this on raspberry pi 4.

Any suggestions ?

thx

`
tesla-control -debug -ble add-key-request rPi6_pub_key.pem owner cloud_key
2023-12-12T09:16:17-05:00 [debug] Set VIN to 'LRWYGDEEOPCxxxxxx'
2023-12-12T09:16:17-05:00 [debug] Set session cache file to ''
2023-12-12T09:16:17-05:00 [debug] Set key name to 'rPi6'
2023-12-12T09:16:17-05:00 [debug] Set key file to '/home/admin/vehicle-command/rPi6_pri_key.pem'
2023-12-12T09:16:17-05:00 [debug] Set OAuth token name to ''
2023-12-12T09:16:17-05:00 [debug] Set OAuth token file to ''
2023-12-12T09:16:17-05:00 [debug] Set keyring type to ''
2023-12-12T09:16:17-05:00 [debug] Set keyring Debug Logging to 'false'
2023-12-12T09:16:17-05:00 [debug] **Skipping private key loading because FlagPrivateKey is not set**
2023-12-12T09:16:17-05:00 [debug] No private key available
2023-12-12T09:16:17-05:00 [debug] Connecting over BLE...
2023-12-12T09:16:17-05:00 [debug] Creating new BLE device
2023-12-12T09:16:18-05:00 [debug] Searching for BLE beacon Saba34e378372xxxxxx...
2023-12-12T09:16:18-05:00 [debug] Connecting to BLE beacon...
Error: failed to find BLE beacon for LRWYGDEEOPCxxxxxx (Saba34e378372xxxxxx): can't scan: context deadline exceeded
`

fanlessfan, 2023-12-12:
Does the vehicle need to be in the specific state to pair?

L480, 2023-12-12:
I'm currently having the same issue on Raspberry Pi Zero 2 W with an external Bluetooth dongle. Dongle is being detected via `hcitool dev` and exclusively accessed by the "connector/ble" Go package. I've disabled the internal interface with `dtoverlay=disable-bt` in `/boot/config.txt`, but still not working.

fanlessfan, 2023-12-12:
what hardware did you use for your "INSUFFICIENT_PRIVILEGES" problem? it seems you connected to the vehicle.

L480, 2023-12-12:
I used the built in UART BLE interface of the Raspberry Pi, which is working. However, the connection was too unreliable that's why I wanted to switch to some external BLE dongle.

sethterashima, 2023-12-13:
You don't need to load a private key for `add-key-request` because the command is authorized using an NFC card. The log line you highlighted isn't cause for concern. It looks like the your Pi found the vehicle over BLE but timed out while connecting; "context deadline exceeded" is how Golang likes to label timeout errors. You could try [increasing the connection timeout](https://github.com/teslamotors/vehicle-command/blob/e1c70d91c2e99014c46830890d0bfb75d44d85fb/cmd/tesla-control/main.go#L153) but it may simply be an issue with the BLE dongle, drivers, or go-ble library.

fanlessfan, 2023-12-13:
Thank you @sethterashima I did try it inside the car. Maybe it's BLE driver issues.

L480, 2023-12-14:
> You don't need to load a private key for `add-key-request` because the command is authorized using an NFC card. The log line you highlighted isn't cause for concern. It looks like the your Pi found the vehicle over BLE but timed out while connecting; "context deadline exceeded" is how Golang likes to label timeout errors. You could try [increasing the connection timeout](https://github.com/teslamotors/vehicle-command/blob/e1c70d91c2e99014c46830890d0bfb75d44d85fb/cmd/tesla-control/main.go#L153) but it may simply be an issue with the BLE dongle, drivers, or go-ble library.
> 
> Trying to pair from inside the car may help if signal strength or interference are issues.

Can confirm that. BLE seems to be more sensitive than expected. After moving my Pi a few meters away from a metal fence, it works as expected. I have no idea why it temporarily worked via the UART interface.

fanlessfan, 2023-12-15:
I believe there is something wrong with the code. it's not reliable. I tried with my raspberry pi 4, raspberry pi zero w, and both are failed with same error message. I also tried on my intel MacBook Pro and it failed many times (both in and out of car). right before I gave up, I gave it a try just outside my car and it just works to let me tap the key card in the centre console. so my MacBook is listed as a key, but after that I got the same error it doesn't matter which command I use (lock, ping, list-keys, etc) and which location I was (in or out). I guess the ble code needs to be optimized.

L480, 2023-12-16:
As @sethterashima mentioned this probably not SDK related. Try to do an `hcidump` during the tesla-control execution and check if your BLE beacon can be found by your host's Bluetooth stack. If this is not the case, it's related to your Bluetooth hardware.

fanlessfan, 2023-12-17:
Hello @L480 . I run there hcidump, but I can't interpret it. I also run the hcitool lescan and found some thing. one of them is 74:B8:39:62:1D:31 S9c710d358225a7a2C, which has similar format of my ble beacon "Searching for BLE beacon Saba34e378372xxxxxx..."

I run this on my raspberry pi zero w. any though?

HCI sniffer - Bluetooth packet analyzer ver 5.66
device: hci0 snap_len: 1500 filter: 0xffffffff
> HCI Event: Command Complete (0x0e) plen 4
    Reset (0x03|0x0003) ncmd 1
    status 0x00
> HCI Event: Command Complete (0x0e) plen 12
    Read Local Supported Features (0x04|0x0003) ncmd 1
    status 0x00
    Features: 0xbf 0xfe 0xcf 0xfe 0xdb 0xff 0x7b 0x87
> HCI Event: Command Complete (0x0e) plen 12
    Read Local Version Information (0x04|0x0001) ncmd 1
    status 0x00
    HCI Version: 4.1 (0x7) HCI Revision: 0x1fc
    LMP Version: 4.1 (0x7) LMP Subversion: 0x2209
    Manufacturer: Broadcom Corporation (15)
> HCI Event: Command Complete (0x0e) plen 10
    Read BD ADDR (0x04|0x0009) ncmd 1
    status 0x00 bdaddr B8:27:EB:DF:A7:51
> HCI Event: Command Complete (0x0e) plen 68
    Read Local Supported Commands (0x04|0x0002) ncmd 1
    status 0x00
    Commands: ffffff03ccffefffffffec1ff20fe8fe3ff78fff1c00040061f7ffff7ff8ffff
              ff3f0000000000000008
> HCI Event: Command Complete (0x0e) plen 4
    Write Simple Pairing Mode (0x03|0x0056) ncmd 1
    status 0x00
> HCI Event: Command Complete (0x0e) plen 4
    Write Inquiry Mode (0x03|0x0045) ncmd 1
    status 0x00
> HCI Event: Command Complete (0x0e) plen 5
    Read Inquiry Response Transmit Power Level (0x03|0x0058) ncmd 1
    status 0x00 level 0
> HCI Event: Command Complete (0x0e) plen 14
    Read Local Extended Features (0x04|0x0004) ncmd 1
    status 0x00 page 1 max 2
    Features: 0x01 0x00 0x00 0x00 0x00 0x00 0x00 0x00
> HCI Event: Command Complete (0x0e) plen 11
    Read Buffer Size (0x04|0x0005) ncmd 1
    status 0x00
    ACL MTU 1021:8 SCO MTU 64:1
> HCI Event: Command Complete (0x0e) plen 7
    Read Class of Device (0x03|0x0023) ncmd 1
    status 0x00 class 0x000000
> HCI Event: Command Complete (0x0e) plen 252
    Read Local Name (0x03|0x0014) ncmd 1
    status 0x00 name 'rPi0'
> HCI Event: Command Complete (0x0e) plen 6
    Read Voice Setting (0x03|0x0025) ncmd 1
    status 0x00 voice setting 0x0060
> HCI Event: Command Complete (0x0e) plen 5
    Read Number of Supported IAC (0x03|0x0038) ncmd 1
> HCI Event: Command Complete (0x0e) plen 8
    Read Current IAC LAP (0x03|0x0039) ncmd 1
    IAC 0x9e8b33 (General Inquiry Access Code)
> HCI Event: Command Complete (0x0e) plen 4
    Write Connection Accept Timeout (0x03|0x0016) ncmd 1
    status 0x00
> HCI Event: Command Complete (0x0e) plen 12
    LE Read Local Supported Features (0x08|0x0003) ncmd 1
    status 0x00 features 0x1f00000800000000 (Link Layer supports LE Encryption)
> HCI Event: Command Complete (0x0e) plen 7
    LE Read Buffer Size (0x08|0x0002) ncmd 1
    status 0x00 pktlen 0x00fb maxpkt 0x08
> HCI Event: Command Complete (0x0e) plen 12
    LE Read Supported States (0x08|0x001c) ncmd 1
> HCI Event: Command Complete (0x0e) plen 4
[... 217 lines cut ...]
    status 0x00
> HCI Event: Command Complete (0x0e) plen 4
    Write Extended Inquiry Response (0x03|0x0052) ncmd 1
    status 0x00
> HCI Event: Command Complete (0x0e) plen 4
    Write Extended Inquiry Response (0x03|0x0052) ncmd 1
    status 0x00

sudo hcitool lescan
LE Scan ...
75:8A:11:C1:3E:AD (unknown)
75:8A:11:C1:3E:AD (unknown)
7B:D3:D4:4E:8B:10 (unknown)
C8:5E:46:4A:3C:8D inlitebt
58:E1:C4:4E:D7:F5 (unknown)
58:E1:C4:4E:D7:F5 (unknown)
69:76:97:16:A9:84 (unknown)
69:76:97:16:A9:84 (unknown)
7B:73:24:01:7D:C3 (unknown)
74:B8:39:62:1D:31 (unknown)
74:B8:39:62:1D:31 S9c710d358225a7a2C
57:DB:38:F4:D2:CD (unknown)
57:DB:38:F4:D2:CD (unknown)
F1:2E:25:21:CE:2D (unknown)
19:40:51:E5:99:56 (unknown)

qja0707, 2023-12-21:
I have same issue, I am testing my macbook pro and doens't work. Also trying to put my public_key.pem in my car's white list with this sdk and tesla app using `https://www.tesla.com/_ak/{my_domain.com}`. Both doesn't work. How can I register my public key to car?

fanlessfan, 2023-12-21:
@qja0707 agreed. it's not hardware issue. it's the code issue.

Jeoffreybauvin, 2024-06-12:
Did someone managed to debug this ?

I have the same issue : yesterday, it was working well. I slept, and now it's not working anymore.

I did not move the bluetooth dongle, nor my server. And the Tesla is at the same spot...

```
root@srv:/usr/local/bin/app# /usr/local/bin/app/tesla-control  -vin LRW.... -ble -key-name /pouet/private.pem -key-file /pouet/private.pem  -debug ping
2024-06-12T21:42:56Z [debug] Set session cache file to ''
2024-06-12T21:42:56Z [debug] Set OAuth token name to ''
2024-06-12T21:42:56Z [debug] Set OAuth token file to ''
2024-06-12T21:42:56Z [debug] Set keyring type to ''
2024-06-12T21:42:56Z [debug] Set keyring Debug Logging to 'false'
2024-06-12T21:42:56Z [debug] Client public key: 044xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
2024-06-12T21:42:56Z [debug] Connecting over BLE...
2024-06-12T21:42:56Z [debug] Creating new BLE device
2024-06-12T21:42:57Z [debug] Searching for BLE beacon S106089xxxxxxxxx...
2024-06-12T21:42:57Z [debug] Connecting to BLE beacon...
Error: failed to find BLE beacon for LRWxxxxxxxxx (S106089xxxxxxxxxx): can't scan: context deadline exceeded
```

thefireblade, 2025-01-22:
A lot of issues has been addressed in https://github.com/teslamotors/vehicle-command/pull/269 closing this old issue for now but re-open a new issue if this persists

closed 2025-01-22 by thefireblade