teslamotors/vehicle-command #104: Can't get the public_key from "/api/1/partner_accounts/public_key"

Coger-yellow, 2023-12-13:
i got my public_key result at the api : "/api/1/partner_accounts"
[image]
but i got the return like this of the api: "/api/1/partner_accounts/public_key"
[image]
if i  use base URL: https://fleet-api.prd.na.vn.cloud.tesla.com
"/api/1/partner_accounts/public_key" returns like this ⬇️
[image]
can‘t figure out which part was wrong.

Bre77, 2024-01-12:
Make sure you registered (POST) your public key to both https://fleet-api.prd.eu.vn.cloud.tesla.com/api/1/partner_accounts/public_key and https://fleet-api.prd.na.vn.cloud.tesla.com//api/1/partner_accounts/public_key.

You will get out of region errors even if the region is correct, if you did not do a registration.

psmedley, 2024-01-26:
I've successfully registered, but aren't  able to query the public_key. Using the syntax suggested  in the API docs:
`curl --header 'Content-Type: application/json' \
  --header "Authorization: Bearer $TESLA_API_TOKEN" \
  'https://fleet-api.prd.na.vn.cloud.tesla.com/api/1/partner_accounts/public_key' 
`
I get:
{"response":null,"error":"param is missing or the value is empty: domain","error_description":""}

If I include:
--data '{    "domain": "string"  }' 

I get a  404 error.

Bre77, 2024-01-27:
> `
> I get:
> {"response":null,"error":"param is missing or the value is empty: domain","error_description":""}
> 
> If I include:
> --data '{    "domain": "string"  }' 
> 
> I get a  404 error.

It's a GET request with domain in the query not the body.

closed 2024-02-14 by sethterashima

nberardi, 2025-01-04:
For anyone running into this issue like I was after migrating to HA 2025.1, here is how I solved the error without having to host the public key myself.  

1. Went to Cloudflare 
2. Went to `Workers & Pages`
3. Clicked `Create`
4. Selected the `Pages` tab
5. Followed the prompts
6. Added my public-key to a zip file that mimicked the path structure Tesla is looking for `/.well-known/appspecific/com.tesla.3p.public-key.pem` for the key.
7. Uploaded the key into the directory created.
8. Then used the domain provided by Cloudflare for the pages `https://myteslacar.pages.dev` as the Allowed Origins in my Tesla Developer Account.
9. Then finally followed Step 4 in the Tesla setup and onboarding by posting 

``` bash
curl --request POST 
        --header 'Content-Type: application/json' \ 
        --header "Authorization: Bearer $TESLA_API_TOKEN" \ 
        'https://fleet-api.prd.na.vn.cloud.tesla.com/api/1/partner_accounts?domain=myteslacar.pages.dev'
```

Please note that `myteslacar.pages.dev ` is an example of a Cloudflare pages domain, you will need to use your own that was generated or you may want to find it to your own domain as something like `tesla.example.com`.