teslamotors/vehicle-command #105: Region for Middle East cars returns NA

rawmean, 2023-12-13:
Region for Middle East cars returns `https://fleet-api.prd.na.vn.cloud.tesla.com` and as a result commands do not work and return an "out of region" error. 

According to documentation, "region" endpoint must return `https://fleet-api.prd.eu.vn.cloud.tesla.com` for Middle East.

joshendy, 2023-12-15:
I'm having the same issue. It seems regardless of user it returns "na" as the region with the NA fleet URL. I was wondering if this is because I didn't opt in for the "Profile Information" (user_data) scope when first setting up app access on Tesla's developers page.

rawmean, 2023-12-15:
> I was wondering if this is because I didn't opt in for the "Profile Information" (user_data) scope when first setting up app access on Tesla's developers page.

I don't think so because the `region` endpoint does not require that "scope". 
The `region` endpoint does return the EU baseURL correctly for users in EU/UK

gshaviv, 2023-12-19:
Getting the same issue, any resolution?

gshaviv, 2023-12-21:
debugging the code, I see the ou_code that is received in the json for the test account I'm testing that is a European account is NA and hence the proxy tries to connect to the North America server and hence gets the error user is out of region. So the issue seems to lie with the wrong ou_code, not sure how that can be fixed or why it's wrong.

zerocooling, 2024-02-18:
Does someone solve that problem ?

thefireblade, 2025-01-22:
It depends on where the vehicle is located but this should just point to the EU endpoint. Closing this old issue as this might vary from a vehicle basis.

closed 2025-01-22 by thefireblade