teslamotors/vehicle-command #107: some tesla accounts return null region

jbanyer, 2023-12-13:
We have seen at least 6 tesla accounts return this response to the region endpoint:

```
GET https://fleet-api.prd.na.vn.cloud.tesla.com/api/1/users/region
{"response":{"region":null,"fleet_api_base_url":null}}
```

The problem seems to go away for some users. Not sure for all users as they haven't retried.

rawmean, 2023-12-14:
Do you know if those users are in NA or EU or somewhere else?

jbanyer, 2023-12-14:
> Do you know if those users are in NA or EU or somewhere else?

Most in NA, one in EU (unconfirmed)

thefireblade, 2025-01-22:
this might mean that their user doesn't exist and they haven't used the mobile app at least once. Closing this issue, feel free to re-open if this is still occurring.

closed 2025-01-22 by thefireblade