teslamotors/vehicle-command #109: Remote Boombox over proxy not working

Urkman, 2023-12-18:
For me the remote_boombox endpoint is not working.
I send the post to my proxy and the logs show just this:

```
2023-12-17T14:33:48Z [info ] Received POST request for /api/1/vehicles/VIN/command/remote_boombox
2023-12-17T14:33:48Z [debug] Executing remote_boombox on VIN
```

When I send another call I get this log:
```
2023-12-17T14:25:47Z [info ] Received POST request for /api/1/vehicles/VIN/command/flash_lights
2023-12-17T14:25:47Z [debug] Executing flash_lights on VIN
2023-12-17T14:25:47Z [info ] Starting dispatcher service...
2023-12-17T14:25:47Z [info ] Session for DOMAIN_INFOTAINMENT loaded from cache
2023-12-17T14:25:47Z [info ] Session for DOMAIN_VEHICLE_SECURITY loaded from cache
[and a lot more]
```

itsMeDavidV, 2023-12-18:
not supported in the proto files i believe. Might just be relegated to fleet api for now.

thefireblade, 2025-01-22:
This should be working, I am closing this issue in favor of a recent report https://github.com/teslamotors/vehicle-command/issues/242.

closed 2025-01-22 by thefireblade