teslamotors/vehicle-command #110: CLOSURE_MOVE_TYPE_OPEN & CLOSURE_MOVE_TYPE_CLOSE don't work in trunk requests

itsMeDavidV, 2023-12-18:
CLOSURE_MOVE_TYPE_OPEN & CLOSURE_MOVE_TYPE_CLOSE requests for trunk return 503 like in issue #56, however unlike in that issue, the trunk does not move at all.

CLOSURE_MOVE_TYPE_OPEN & CLOSURE_MOVE_TYPE_CLOSE are needed, bc using CLOSURE_MOVE_TYPE_MOVE isn't always ideal, especially when the trunk is halfway open/closed, depending on the user's perspective. It would be counter intuitive to not let them chose a direction in that case.

sethterashima, 2023-12-18:
Can you test against https://github.com/teslamotors/vehicle-command/pull/108/files?

Note the documentation changes on OpenTrunk/CloseTrunk/MoveTrunk; some cars lack the hardware for powered open (as opposed to unlatch) and close.

itsMeDavidV, 2023-12-19:
Hi @sethterashima sorry I should note that I wasn't using the SDK when this error occurred, I called `signed_command` directly. 

I also replicated the exact same behavior on [your branch](https://github.com/sethterashima/vehicle-command/tree/fix-503-and-retry) just now as well as on the [`v0.0.2` release.](https://github.com/teslamotors/vehicle-command/releases/tag/v0.0.2)

`OpenTrunk()` & `CloseTrunk()` don't open/close trunk and 503 is returned.

Should also note, when using the SDK, I don't have the proxy set up rn, just called said commands directly, so wasn't able to test automatic retires that proxy would implement, though manual retries don't have no affect on outcome either.

> some cars lack the hardware for powered open (as opposed to unlatch) and close.

Tested on M3 21 so can confirm the vehicle does have open/close actuation hardware & sw capability

thefireblade, 2025-01-22:
Not sure if this is still an issue, 503 generally indicates that the vehicle is offline or is refusing to come online. I will close this old issue but feel free to re-open a new issue if this is occurring

closed 2025-01-22 by thefireblade