teslamotors/vehicle-command #111: Open-source / self-hosted applications

MikeBishop, 2023-12-20:
In order to get a developer key, you must register as a "business entity," which I interpret to mean individual developers are excluded. Also, the current authentication setup appears to exclude open source applications which are run by individual users -- each installation retrieves data for that user's vehicles, and the software has no central server endpoint which will relay all requests back to Tesla.  For open source software, it's obviously impossible to have a secret which is known only by the developer and Tesla, since the secret will have to be in the code for each user to run.

What is the proposed path forward for such self-hosted applications?  Can users obtain OAuth tokens for their own accounts/vehicles without needing a developer account per user?

sethterashima, 2023-12-22:
Feel free to put "Personal Use" or similar in any Business Name fields. Individual developers are more than welcome to enroll.

We're having some internal discussion on the best way forward for self-hosted applications, including discussions about providing an endpoint for obtaining personal-use Fleet API tokens without a developer account or dedicated domain. We don't want to force self-hosted apps to migrate to a centralized model, but we also want to avoid inadvertantly incentivizing centralized apps to solicit user passwords or personal-use Fleet API tokens.

SebastianGode, 2023-12-22:
Hey @sethterashima 
Thanks for this answer.
What should I enter as TAX-ID in the Developers registration formular?
[image]
It is required, but simply entering "DE123456789" to match the RegEx Pattern doesn't work and gives me an error on submitting the final form.

Also the Fleet-API support wrote me several weeks ago, that I'm required to have a business. Has this been changed now?

MikeBishop, 2023-12-27:
> We're having some internal discussion on the best way forward for self-hosted applications, including discussions about providing an endpoint for obtaining personal-use Fleet API tokens without a developer account or dedicated domain.

Can I suggest not removing the API we're all currently using until you've concluded these discussions and given our apps a path forward?

I'd think some kind of limit that makes it impractical for one service to use many personal-use tokens would work. Not accepting multiple personal tokens on a single connection, or limiting the number that can be used from a single source address might be reasonable (though tread carefully around CGNATs with the second).

andig, 2023-12-27:
I was wondering the same and asked myself, if we as open source project ([evcc.io](https://evcc.io)) could share our ClientID as part of the code? Using our client id, users could- given they have a token retrieved through the same client id- send query and wake-up, but not send commands to, their cars. Sending commands would still require the private key which we'd obviously not open source.

Would that create and security risks for our users?

SebastianGode, 2024-01-07:
Hey @sethterashima @agbpatro 
I tried getting my question above answered by contacting the E-Mail support fleetapisupport@tesla.com but I sadly haven't got an answer.
Would it be possible for you to clarify the following:

- How does a registration to developer.tesla.com work as a private consumer? (as we can't enter a business tax id)
- Will there be a local Wifi-based API available in the future to circumvent the API limits from Tesla as long as the server and car are in the same network? So something like the current BLE API, just through Wifi.

Are there maybe also any other news regarding this whole topic?

MikeBishop, 2024-01-22:
> We're having some internal discussion on the best way forward for self-hosted applications, including discussions about providing an endpoint for obtaining personal-use Fleet API tokens without a [...]

While PRs exist for several of the apps I use to switch to the Fleet API, the current system still requires each user to have a Developer API registration, a publicly accessible endpoint with keys, etc. That's easy enough for _me_ to set up, but not exactly for all my users.

Given that we're closing in on the end of January, we'd all really like to hear what the proposed path forward is for home users who aren't set up for these sorts of things.

andig, 2024-01-24:
For time being we (https://evcc.io) have stopped shipping software for sending commands to Tesla vehicles over fleet api. Data retrieval still seems fine with CLIEND_ID bundled with pre-built application, but distributing the CLIENT_SECRET too seems a no-go.

jbanyer, 2024-01-24:
> We're having some internal discussion on the best way forward for self-hosted applications, including discussions about providing an endpoint for obtaining personal-use Fleet API tokens without a [...]

FWIW, as the author of a centralised commercial app [https://chargehq.net], if Tesla explicitly stated that personal-use tokens were prohibited for use by commercial apps, I would respect that. You can't (or shouldn't) build a commercial product based on an API integration which is explicitly prohibited.

MikeBishop, 2024-01-30:
For now, I've [created a Node app](https://github.com/MikeBishop/fleetapi-tokens) which self-host users can use to obtain Fleet API tokens for their own use.  Obviously, it still requires a domain name, a developer account, and an app registered on the Developer portal.

An official way to obtain single-account tokens without developer registration would, of course, obviate the need for this and is preferable for those of us self-hosting.

llamafilm, 2024-04-05:
I’ve created a [Home Assistant addon](https://github.com/llamafilm/tesla-http-proxy-addon) that runs the API proxy, and helps simplify the authorization flow for self-hosted use. I worked around the domain requirement with a hacky approach: When the callback leads to a non-existent URL, the user copies the code from that URL and then proceeds to get a token. The “Auth for Tesla” iOS app does something similar.

So the domain is really only required for hosting the public key, and I’ve found that a free DuckDNS domain does work.

Still, it would be great if Tesla could support this in some official way, by hosting the public key on Tesla’s server and build an auth flow that doesn’t require a callback.

jholster, 2024-05-25:
I am individual developer / hobbyist / consumer who would like to self-host the above mentioned projects. Would be great if there was official way to register as private developer. Currently it's not possible because TAX ID is required.

Abcmsaj, 2024-06-09:
I'm sorry but it's absolutely insane that the old API has been depreciated and little niggles like this are still being worked out. I am now unable to control the charging on my car from HomeAssistant because I lost access to the old API a few days ago (surprised I managed to get 6 months extra!) and now I can't make a developer account to use the new Fleet API because I don't have a valid UK Tax ID...

SebastianGode, 2024-06-10:
@Abcmsaj You might want to have a look at https://github.com/pkuehnel/TeslaSolarCharger which also works through BLE and does not require any API access. It also works though the new fleet-API but that's limited to 50commands per day since last week.

sethterashima, 2024-08-07:
https://developer.tesla.com/docs/fleet-api/authentication/open-source-tokens

closed 2024-08-07 by sethterashima

MikeBishop, 2024-08-07:
@sethterashima, this is a great step!  Can you clarify one thing, though. It appears that the URLs still have to be specified during registration, and those might vary per environment for self-hosted applications. How is this intended to work?

SeanLMcCullough, 2024-08-28:
Is my understanding correct that if one is to set up an open source account that it is currently impossible to use the new Vehicle Commands API now that REST is deprecated? It seems in order to use the commands the workflow of setting up keys is required which is not possible with an open source token. It also seems impossible to use client_credentials to get a partner token as there is no client_secret. Calling the REST command endpoints all return HTTP 403 responses for my 2024 M3.

It seems that in order to set up to trust my app's public key, I have to perform the partner registration endpoint. `POST /api/1/partner_accounts`.

> https://developer.tesla.com/docs/fleet-api/authentication/open-source-tokens

I have everything else in place but calling this with my open source token yields :

```
{"error":"the register endpoint requires a partner authentication token, see https://developer.tesla.com/docs/fleet-api/authentication/partner-tokens (401)"}
```

I suppose in this case I have to register as a personal-use non-open-source project?