teslamotors/vehicle-command #112: Please add heater off and AC off functions to api

beachmiles, 2023-12-21:
A functions are missing in the HVAC API  along with the car GUi.
No vent mode
No heater off mode. 
The hack for the past 4 years plus for both of these cases is to set the temp to LO and turn off the AC button which is ridiculous.

In the meantime can we please add to the API aaa way to toggle the AC button on or off, instead of only allowing us to turn on or off the entire HVAC.

Ideally there would be a full featured thermostat in the car with 2 temp setpoints, the heater on temp and the AC on temp and any temp in between is vent mode.
Thanks.

thefireblade, 2025-01-22:
The API already has this through https://developer.tesla.com/docs/fleet-api/endpoints/vehicle-commands#auto-conditioning-stop. This is an old issue so closing for now

closed 2025-01-22 by thefireblade

beachmiles, 2025-01-24:
> The API already has this through https://developer.tesla.com/docs/fleet-api/endpoints/vehicle-commands#auto-conditioning-stop. This is an old issue so closing for now

The API does not have a heater off or vent only mode command while you are driving through vehicle. Neither does the display inside the car.

You referenced an the API command auto_conditioning_stop which turns off all climate preconditioning heater and AC. 

The only way to cool the car with AC and not have the cabin heater constantly running is by putting the temp to LO. I have verified this behavior on almost every build for 4 years now using scanmytesla along with a DC current clamp meter directly on the high voltage line going to the PTC cabin heater.

Ideally you could also have a vent mode where no AC or heating happens and only the fan blows, and/or having 2 temperature setpoints like a home thermostat and the range between the low and high setpoints is fan only.